rednblkx/HomeKey-ESP32 #1: HomeKey Authenticating Issue

ihrapsa, 2024-01-06:
First of all  thank you and congrats for putting this together!

I managed to get the project working on an ESP32 with PN532 connected through SPI and I've paired it with HomeKit. The homekey is added to wallet, the "dummy" switch works, however **the key does not seem to toggle the switch**. Phone responds to the ECP by selecting the key but the authentication does not succeed, ending with a red (!) mark saying **"Try again"**. I tried manually selecting the key but nothing changes.

[image]

This is the last part of the logs on the serial monitor:

```
[...]
Response Status: 91
 DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF  DE AD BE EE FF DE AD BE EE FF 

SHARED KEY:   DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE
DERIVED KEY:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE

DATA MATERIAL:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DEDE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE

KEYING MATERIAL:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE

DATA MATERIAL:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DEDE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DEDE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE

KEYING MATERIAL:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE
PERSISTENT KEY:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE
VOLATILE KEY:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE
KENC:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE 
KMAC:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF DE 
KRMAC:  DE AD BE EE FF DE AD BE EE FF DE AD BE EE FF
 DE AD BE EE FF DE AD BE
DE AD BE EE FF DE AD BE

decrypted data: 
decrypted length: 0

SOMETHING WENT WRONG
 80 3C 00 00

Response Status: 3
```
Any idea what might be wrong, or is this not implemented yet?

Thank you!

rednblkx, 2024-01-06:
Hi, thank you for checking out this project, glad to see someone is already interested in this.

It seems that something went wrong during the key derivation here for the STANDARD auth here [main.cpp#L1075](https://github.com/rednblkx/HomeKey-ESP32/blob/97eae7e0962c70d9225cde3352755d6d8fe085bb/src/main.cpp#L1075)

As you said that ECP works it means the lock knows the same identifier as the phone and therefore it was also provisioned with the private key, so that's one down.

And the other keys used at that stage are generated on the ESP except the phone's public key that you get here [main.cpp#L1284](https://github.com/rednblkx/HomeKey-ESP32/blob/97eae7e0962c70d9225cde3352755d6d8fe085bb/src/main.cpp#L1284) so the issue might be there.

I'm not able to replicate the issue so i'll need some more logs from you to get to the bottom of this, you can delete all the actual data, right now i only need to inspect the length of the said data which would be printed as "Response Status", "{Something} length" or for TLV data you would have like *** TLV TAG 4d[16] with the tlv tag followed by length in parentheses, i know it's a mess :weary:

Please let me know if you get any errors or crashes throughout the process, the code is kinda rough at the moment so it's not exactly "safe".

ihrapsa, 2024-01-06:
Here are the log bits requested:

```
Response Status: 9
selectCmdResLength: 8
```

```
*** TLV TAG 4d[16]: 
DE AD BE EF DE AD BE EF DE AD BE EF DE AD BE EF
*** TLV LENGTH: 
107
```

Also, while compiling I kept getting this error on each 'cmac_ctx' occurance:

```
src/cmac.c: In function 'mbedtls_cipher_cmac_starts':
src/cmac.c:216:10: error: 'mbedtls_cipher_context_t' {aka 'struct mbedtls_cipher_context_t'} has no member named 'cmac_ctx'; did you mean 'cipher_ctx'?
     ctx->cmac_ctx = cmac_ctx;
          ^~~~~~~~
          cipher_ctx
```

I managed to fix this by doing what the error said (replace 'cmac_ctx' with 'cipher_ctx'). That 

This in turn created the following error 

```
src/cmac.c: In function 'mbedtls_cipher_cmac_update':
src/cmac.c:238:28: warning: dereferencing 'void *' pointer
     state = ctx->cipher_ctx->state;
                            ^~
src/cmac.c:238:28: error: request for member 'state' in something not a structure or union
```
This one got butchered and fixed by replacing 'state = ctx->cipher_ctx->state;' with 'state = ctx->cipher_ctx;'

After all this I managed to build and upload

I reckon my issue has something to do with my errors

That'll be all 😵

rednblkx, 2024-01-07:
well there's our issue then, the last 8 bytes of the encrypted response you get from the device here [main.cpp#L1070](https://github.com/rednblkx/HomeKey-ESP32/blob/97eae7e0962c70d9225cde3352755d6d8fe085bb/src/main.cpp#L1070) are a message authentication code(MAC) that the receiver can use to assure the authenticity of the message and we do that here [DigitalKeySecureContext.cpp#L127C5-L130C73](https://github.com/rednblkx/HomeKey-ESP32/blob/97eae7e0962c70d9225cde3352755d6d8fe085bb/src/util/DigitalKeySecureContext.cpp#L127C5-L130C73) with the cmac function and we compare the result with the with what we received.

the cmac.c, alignment.h and common.h were copied from the mbedtls library as i'm using platformio with the vscode extension at the moment and the version of esp-idf that's included does not have mbedtls with the cmac functions enabled so i just included them in this project for the time being to not waste time on configuring platformio and whatnot. 

Question is, are you using PlatformIO? and what version?
You could try the platformio that's bundled in the vscode extension if you haven't already or run it in a docker container though this can a double-edged sword.

I'll look into getting rid of those local dependencies.

rednblkx, 2024-01-07:
in the attempt of solving this, i stumbled upon your, it seems that my current solution only works with the precompiled esp-idf that's included on platformio when using the arduino framework.

i suggest to use platformio with whatever it defaults.

ihrapsa, 2024-01-07:
I've been using PlatformIO IDE extension (v3.3.2, core v6.1.11, home v3.4.4) from VSCode all along. I've also tried starting from scratch with VSCode on ubuntu with default platformIO settings but the same errors occur.

rednblkx, 2024-01-07:
think i found the issue, the files i copied over are still using stuff from the mbedtls library which are under the condition that MBEDTLS_CMAC_C is defined, so i added a build flag to platformio.ini which "enables" CMAC

rednblkx, 2024-01-07:
it seems that it was working on my environment because i also have the linux version installed globally which is different than the one written for the ESP

ihrapsa, 2024-01-07:
Yup. It's working now! 🤩 
Thanks a lot! 🙂 
I think we can close this one.

rednblkx, 2024-01-07:
Great to hear and let me know if anything, I’m open to suggestions.

closed 2024-01-07 by rednblkx