rednblkx/HomeKey-ESP32 #5: Authentication issue

rednblkx, 2024-01-19:
"it seems the new code didn't worked for me - I´ve tried it twice (with an iPhone 14 and Apple Watch 8). Reseted everything - but still not working (log shows plenty of stuff - but no success)"

_Originally posted by @thorsten-gehrig in https://github.com/rednblkx/HomeKey-ESP32/discussions/3#discussioncomment-8160842_

thorsten-gehrig, 2024-01-20:
Additional info (may relevant for troubleshooting): the home does have a family connected with 5 Apple-ID - including 3 Apple watches, 7 or 8 iPhones and 3-4 iPads. Hence there are various devices which can "use" the HomeKey automatically.
Not sure if this does have any impact?

rednblkx, 2024-01-20:
That might be a factor, however that should only be speed issue not straight fail, as in order for the "FAST" flow to work, it just loops over all the issuers(unique per AppleID) and endpoints(devices) of each issuer to check if they are match for the cryptogram the lock receives from the device and if none matches then it fallbacks to the STANDARD flow which is slower by comparison to the fast flow but overall faster as it doesn't do calculations for every endpoint of every issuer.

So first let's try going straight to the standard flow, use command `@F1` to do that and also `@LV` to get more detailed logs to better understand what is going on.

when sending the logs, just halve the hex data so that way the private parts are redacted and i can see some continuity on the logs.

rednblkx, 2024-01-27:
Issue closing, solved in related #8.

closed 2024-01-27 by rednblkx