rednblkx/HomeKey-ESP32 #8: Crash?

thorsten-gehrig, 2024-01-26:
Hi
didnt realize that this is why it´s propably often not working.
Also the iPhone is often not triggering the "apple value add" thing which triggers the HomeKey to be presented.
Does this panic does help in any directon?

`I (3797) HKAuthCtx: Ephemeral Key generated
E (4025) HKAuthCtx: Response Status not 0x90, something went wrong!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d45de  PS      : 0x00060d30  A0      : 0x800e557e  A1      : 0x3ffc42a0  
A2      : 0x3ffc4418  A3      : 0x3ffc4458  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000fb9  A7      : 0x3f405164  A8      : 0x800d45da  A9      : 0x3ffc4280  
A10     : 0x00000000  A11     : 0x3ffc42fc  A12     : 0x00000004  A13     : 0x3ffc4320  
A14     : 0x3ffc42f0  A15     : 0x3ffc434c  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x400d45db:0x3ffc42a0 0x400e557b:0x3ffc4400 0x400f563f:0x3ffc45a0 0x400f5747:0x3ffc4670 0x400d4d76:0x3ffc4690 0x4011791d:0x3ffc46b0

  #0  0x400d45db:0x3ffc42a0 in AuthenticationContext::fast_auth(bool) at src/auth/authContext.cpp:415
  #1  0x400e557b:0x3ffc4400 in LockMechanism::loop() at src/main.cpp:179
  #2  0x400f563f:0x3ffc45a0 in Span::pollTask() at .pio/libdeps/wemos_d1_mini32/HomeSpan/src/HomeSpan.cpp:278
  #3  0x400f5747:0x3ffc4670 in Span::poll() at .pio/libdeps/wemos_d1_mini32/HomeSpan/src/HomeSpan.cpp:164
  #4  0x400d4d76:0x3ffc4690 in loop() at src/main.cpp:871
  #5  0x4011791d:0x3ffc46b0 in loopTask(void*) at /Users/thorsten/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 30765105d23cddf3

Rebooting...
ets Jul 29 2019 12:21:46
`

rednblkx, 2024-01-26:
Hi,

The crash occurred due to an oversight from my part, it will be fixed, however, it only got there because the device didn’t agreed with what was sent as can be seen by the error before the crash.

Could you please provide a debug level log by using `@LD` to get more context? Also do you have express mode on? And could clarify what is the "apple value add" thing?

thorsten-gehrig, 2024-01-27:
Thanks.
I will create the logs later
With "apple value add" thing i meant the VAS and ECP 
=> https://github.com/kormax/apple-vas
=> https://github.com/kormax/apple-enhanced-contactless-polling
- or in other words: the home-key didnt show up in many cases. Sometimes it showed up - but didnt authenticate successful... and rarely i was able to authenticate.

Ill post fresh logs later. Thanks for all the effort!
Best
Thorsten

rednblkx, 2024-01-27:
Ok, great.

So is this also what was happening a couple days ago as mentioned in #5 ?

From the behavior you are describing, there might possibly be an issue with the PN532 unit, consider trying another one, however, you might know that there’s an issue with PN532 clones that use cheap resistors and stuff, kormax also talked about this in his ECP repo.

Unfortunately, the library for the PN532 does not currently implement a way to identify if there is an RF I/O issue like it is with nfcpy in Python which is something I’ll deal with at some point among other things.

thorsten-gehrig, 2024-01-27:
Some updates:
I've replaced the PN532 - and it works much better.
With the watch almost every time - on my phone i was still struggling to identify the HomeKey and showed "PASSIVE TARGET SELECTED". After a while i figured out i had the express mode also activated for a travel card (Had "Clipper card" enabled - https://www.clippercard.com/ClipperWeb/apple-pay ). After disable express mode for Clipper card i have the same high success rate on the iPhone.
I have still here and there
`I (1041693) HKAuthCtx: Ephemeral Key generated
E (1041817) HKAuthCtx: Response Status not 0x90, something went wrong!
W (1041875) LockMechanism: Authentication Failed, lock state not changed`
But also a good success rate.
Do you need the same @LD logs as before - or do you see the same behave on your side?

MQTT does also works fine... (a BIG thank you!)  so I'm very close to replace my existing "stupid" NFC reader at my door with this new one :-) :-)

Best
Thorsten

rednblkx, 2024-01-27:
Awesome!
If things work for you now, no logs are needed anymore also I haven’t been able to replicate this behaviour so you might still have some issues with the module, faulty connections maybe?

closed 2024-01-30 by rednblkx