rednblkx/HomeKey-ESP32 #10: [Upload issue]: overlap at address: 0x1000

ihrapsa, 2024-01-28:
It builds fine but uploading fails with:

```
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x1000 for file: ~HomeKey-ESP32/.pio/build/release/bootloader.bin
```
I built it with the project fresh/clean. PIO VSCode add-on is latest version.
I'm using an ESP32 dev module.

rednblkx, 2024-01-29:
This is giving me a headache as it doesn't make any sense.

Did you tried erasing the flash before uploading?

Also, try to do a verbose upload from PIO and let me what arguments were used for esptool

the command should be something like this
```
"/home/red/.platformio/penv/bin/python" "/home/red/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32 --port "/dev/ttyUSB0" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 /home/red/Documents/PlatformIO/Projects/HomeKey-ESP32/.pio/build/release/bootloader.bin 0x8000 /home/red/Documents/PlatformIO/Projects/HomeKey-ESP32/.pio/build/release/partitions.bin 0x19000 /home/red/Documents/PlatformIO/Projects/HomeKey-ESP32/.pio/build/release/ota_data_initial.bin 0x20000 .pio/build/release/firmware.bin
```

ihrapsa, 2024-01-29:
Did multiple flash erases. Will try a verbose upload as soon as I get home. Will try on my Linux system as well (this was on macOS M1). Will update progress here.

ihrapsa, 2024-01-29:
Ok, it seems to be related to my environment despite not changing anything from previous versions. Can't tell what causes it but on a Linux x86 machine it builds and uploads just fine. However, it doesn't generate the homekey this time, only the dummy lock works:

```
DEVICE NOT YET PAIRED -- PLEASE PAIR WITH HOMEKIT APP

I (10388) wifi:<ba-add>idx:0 (ifx:0, 60:32:b1:16:83:f2), tid:0, ssn:0, winSize:64

*** ERROR: Unrecognized Controller PairingID

AQEBBgA=
I (56018) NFCAccess: GET READER KEY REQUEST
AQEBBgA=
I (56115) NFCAccess: GET READER KEY REQUEST
AQECBi8BAQI..................................................wizXCy5XVlSiw==
I (56505) NFCAccess: SET READER KEY REQUEST
I (56684) NFCAccess: KEY SAVED TO NVS, COMPOSING RESPONSE
I (56685) NFCAccess: RESPONSE LENGTH: 8, DATA: BwMCAQA=
AQECBi8BAQI.........................................MPgAwizXCy5XVlSiw==
I (56785) NFCAccess: SET READER KEY REQUEST
I (56961) NFCAccess: KEY SAVED TO NVS, COMPOSING RESPONSE
I (56962) NFCAccess: RESPONSE LENGTH: 8, DATA: BwMCAQA=

*** ERROR: Unrecognized Controller PairingID
``` 

And `@P` shows: 

```
@P
I (164129) print_issuers: HOMEKEY ISSUERS: 0
```
😩

rednblkx, 2024-01-30:
Glad it finally worked, i figure esptool might be doing some weird stuff on macOS though i did some changes in the meantime, don't know if it actually made a difference.

As for the homekey data, that is dependent on the device to provision this during setup.
On the lock alone we can only generate the id of each issuer from their HAP public key and this happens whenever a device makes a provision request.

Although, the data can also be provisioned during the NFC Attestation flow but that is not currently implemented as i've removed it during cleanup due to being too convoluted and not exactly required but will be brought back sometime soon.

Still, weird you got zero device credential request, make sure that you finish the setup in order for the device to send the provision request.

ihrapsa, 2024-02-03:
Had to refresh entire environment to get rid of that upload issue. I think we can close this one.

closed 2024-02-03 by ihrapsa