rednblkx/HomeKey-ESP32 #11: Possible?

tonysprenk, 2024-01-31:
Is it possible to use this firmware to create an integrated HomeKit lock using this hardware: 

https://a.aliexpress.com/_EHdQUiD
https://a.aliexpress.com/_EJBjFXR

what would be the steps to make this work? 

Thanks!

ihrapsa, 2024-01-31:
Should work on that esp32 relay module, although the relay control part logic should be added into code on successful authentication. HomeSpan has some nice tutorials on that afaik. Make sure you get the PN532 module (red).

tonysprenk, 2024-01-31:
Nice. I’m mostly familiar with [HAA](https://github.com/RavenSystem/esp-homekit-devices) so hopefully I’ll be able to work with your implementation. The dream would be to have your code integrated into HAA.

rednblkx, 2024-01-31:
HomeSpanUI looks very nice, however, i may move away from HomeSpan in the future as i don't see it as a good fit anymore, but that idea is still brewing so it may not come to fruition.

Regarding the topic of this discussion, GPIO Control is not planned, but anyone interested can just replace the `mqtt.publish` lines with whatever they want. The HomeSpan library does have a number of example, not sure if any of them are for a relay, however, that should only be a matter of setting the pin HIGH or LOW for open or close whenever needed.

rednblkx, 2024-02-04:
Closing Issue, this can be more extensively discussed on #17

closed 2024-02-04 by rednblkx