rednblkx/HomeKey-ESP32 #12: ESPHome

Juggler00, 2024-01-31:
Any chance you will be able to integrate your project with [ESPHome](https://www.esphome.io)? There are thousands of Homeassistant users that already have ESPHome integrated with their deployments – adding an ESP32-based HomeKey lock would be amazing!!!

rednblkx, 2024-01-31:
that is the plan, though it will be a separate repository and i'll have to first develop a HAP component for ESPHome in order for all of this to work.

Juggler00, 2024-02-01:
That is amazing! My though is to use a PoE powered ESP32… would be perfect!

chbmuc, 2024-02-01:
I must admit, that I've not completely understood the message flow for Homekit/Homekey, so this is just an idea:

I guess developing a HAP component for ESPHome would be quite an effort (although this would be interesting for other use cases as well). So, why not keep the Homekit code in Homeassistant and just extend the API to forward the NFC-messages from ESPHome to Homeassistant. HAP-python already seems to have basic Homekey support - (ikalchev/HAP-python#454).

davidteles, 2024-02-01:
I would love to have at least an easy way to have an entity on Home assistant from this. So that way I could have the esp just focus on the authentication and all the automation of unlocking the door handled by homeassistant. No need to go over the top I would be happy with a simple entity that I could use to trigger the unlocking nothing else

rednblkx, 2024-02-04:
@chbmuc that of course could be done, however, i'm rather looking to develop an all in one package even though i believe most ESPHome users would also be using HA. Also, i just want to try making a HAP component because i know it will be useful in some other cases and would be nice for ESPHome to have native HomeKit.

rednblkx, 2024-02-04:
@davidteles it's in the plan to implement mqtt autodiscovery, but for the time being the current implementation of mqtt can be used for that.

see the [README](https://github.com/rednblkx/HomeKey-ESP32/blob/main/README.md) to configure MQTT.

Note that MQTT_STATE_TOPIC will always receive the target state when the lock is actioned from the Home App and when triggered by homekey auth the inversed state of the lock will be published.

If you wish i can provide you with the config for a simple automation.

budisek, 2024-02-16:
It would be nice to have some easy way to implement it to ESPHome 🙏

idiamant, 2024-04-18:
That will be awesome! I have implemented a smart lock mechanism using esphome. I have an electric lock with dry contact to get status and change state (lock/unlock).
Looking to implement Homekey support too for that lock, and having an integration for esphome and homekey would make it much more intuitive (and spare me from having two lock entities at my Home app 😀)

sven253, 2024-08-03:
i will test it in a few days. I ordered new hardware for testing. By the way Elechouse has an new PN532 modul "PN532 NFC RFID module V4" anyone tried this one? I am new to to esphome can you provid a basic yaml for testing maybe just a lock or an automation to turn on a light on unlock.

rednblkx, 2024-08-03:
> By the way Elechouse has an new PN532 modul "PN532 NFC RFID module V4" anyone tried this one?

Haven't got one but supposedly the V4 has a better antenna since that's mainly the only thing they changed, still uses the same PN532 chip hence the name so software-wise won't have any issues.

> I am new to to esphome can you provid a basic yaml for testing maybe just a lock or an automation to turn on a light on unlock.

That is already included in the example yaml files, [`homekey-test-esp32.yaml`](https://github.com/rednblkx/HAP-ESPHome/blob/main/homekey-test-esp32.yaml) and [`homekey-test-c3.yaml`](https://github.com/rednblkx/HAP-ESPHome/blob/main/homekey-test-c3.yaml) represent the minimal setup to get homekey going but and with all its options showcased.

The [`test.yaml`](https://github.com/rednblkx/HAP-ESPHome/blob/main/test.yaml) file showcases the full functionality of HAP-ESPHome

> Is it possible to use the same device as HomeKey and ESP32 Bluetooth Low Energy Tracker Hub? This would be cool for example autounlock

Don't see why not, it's esphome and the esp32 should have enough power to handle it

rednblkx, 2024-08-03:
Also note for everyone, express mode should now be functional, anybody interested, please give it a go and let me know how it goes

pietro4711, 2024-08-03:
Great news!! I will test it tomorrow.

closed 2024-08-04 by rednblkx