rednblkx/HomeKey-ESP32 #14: 1 out of 2 home devices fail to use express mode

ihrapsa, 2024-02-03:
The latest build has issues when polling the HomeKey on my **iPhone 12 Pro iOS 17.0**. My partner's **iPhone 15 Pro Max iOS 17.3** has no issue with Express mode. I tried pairing it from both of our devices. I erased ESP32 flash multiple times. I rebooted phone, but issue persisted.

Looking at the logs I get:
at the beginning when pairing the lock:
```
Starting HAP Server on port 80 supporting 9 simultaneous HomeKit Controller Connections...

DEVICE NOT YET PAIRED -- PLEASE PAIR WITH HOMEKIT APP
[...]
*** ERROR: Unrecognized Controller PairingID
```

Then when approaching the reader I get:

```
I (284881) LockMechanism: *** PASSIVE TARGET DETECTED ***
```
This is the same behavior as when I'm approaching a random NFC card. However, using the HomeKey in the Wallet the authentication succeeds.

Any idea what's going on? 
I can't seem to change the log level. It said it's on Level 2 but I see no difference from level 0. I'm starting to wonder if the issue is with my phone, however Express Mode works in other circumstances (with transit cards)

rednblkx, 2024-02-04:
This might be caused by the filter i added to identify a homekey early as i've seen that all devices be it an iPhone or a Watch have ATQA 0x04 and SAK 0x20 which thinking about it now is not exactly needed since sending a SELECT command for the HomeKey Applet is enough and based on the response we can rule out the target as homekey.

ihrapsa, 2024-02-04:
Latest dev branch fixes this. Awesome! 😎 
I guess we can close this issue.

rednblkx, 2024-02-04:
Brilliant 🥳, glad that helped. 
Note that you can just ignore the “Unrecognized Controller PairingID”, that’s just a device that is trying to connect to the accessory with the wrong authorization, that might happen when you add a device after you previously removed it or just a device in home that hasn’t paired yet.
Also logs are a bit segregated at the moment, HomeSpan has it’s own log system and I’m just using the integrated logging from esp-idf for all the home key stuff and I added the command `@L` followed by the level name to change log level (N,I,D or V).

PR #15 (2024-02-05): Merge dev into main
- d25ad90410b631a5f8436dd9bd700f1fbced02a6 hopefully fixes #14 

- restructured authentication logic, the auth flow is now entirely handled within the auth context and in main we only get the final result including the Endpoint ID, Issuer ID and the flow that was used(kFlowFAST, kFlowSTANDARD or kFlowATTESTATION) or kFlowFailed if auth failed.
- removed the static hostname suffix since it will then conflict with other devices that use the same hostname and because of this PIO now has to be configured with the hostname in order for OTA to work
- swapped copy semantics with m [... cut]
closed 2024-02-04 by rednblkx