rednblkx/HomeKey-ESP32 #21: Not publishing RFID Cards

Z-pi, 2024-02-11:
Thank you for making this awesome project, it's working perfectly with my Apple devices and publishing the data to MQTT. I just have one problem the esp32 is not publishing the UID or any other information when reading a passive card. The log shows the UID but it is not published to the MQTT "Auth" topic. Is there something I need to activate to get this to work?
[image]
[image]

rednblkx, 2024-02-11:
yeah, i messed on conditionals, i pushed some changes now, please let me know if this fixes it

Z-pi, 2024-02-11:
Thank you! that fixed it, Suggestion for HASS users that want tags to appear in native HASS tags, send the below json to homeassistant/tag/voordeur/front-door-rfid/config 
`{
  "topic": "topic/auth", 
  "value_template": "{{ value_json.uid }}",
  "device":{
    "name":"Voordeur RFID",
    "identifiers":[
         "frontdoorrfid"
       ]
    
  }
}`

closed 2024-02-11 by Z-pi

rednblkx, 2024-02-11:
That’s cool, might use it, thank you. Never really bothered to read how tags work in HASS. I should create a wiki at some point to add stuff like this.

commit 5940f35a (2024-02-11): should fix #21