rednblkx/HomeKey-ESP32 #23: HKAuthCtx: fast_auth > FAST Flow failed!

thorsten-gehrig, 2024-02-14:
Hi
i´m meanwhile have new NFC readers (different brands) and multiple ESPs...
Unfortunaintly the new code always shows me "HKAuthCtx: fast_auth > FAST Flow failed!"
Is there anything i can share which helps to troubleshoot this?

It had earlier not worked perfect (code till EOY23) - but with the new code it seems never working.
Is this maybe related to multiple Homekit-controllers (2 AppleTV and 1 iPod mini)

Thanks for any advice
Kind regards
Thorsten

thorsten-gehrig, 2024-02-17:
It´s somehow strange: I've added a new home - and just added the Homekey. There is no bridge for this home.
it works like a charm and within a fast reaction time - on 2 phones. Works every time.
This may sounds like the "regular Home" is either somehow broken or overloaded (there are around 70 devices & sensors). Or maybe there is an issue as i have 3 bridged (2 AppleTV and 1 HomePod mini)?
BUT (there is always a but...) - it doesn't work on the AppleWatch. It fails every time.
I´ve also see when using Apple watch the console always shows an Auth1 (after the Auth0). The Phones to most time only have Auth0

Any idea how to continue troubleshooting?
Best
Thorsten

kormax, 2024-02-17:
It fails on a watch because ATTESTATION flow hasn't been implemented, and Apple Watch, unlike an iPhone, does not provision it's own key into a lock until you open the Home app and interact with the device for a bit.

thorsten-gehrig, 2024-02-17:
Thanks for the feedback. It helps!
This leaves it back to the question why i fail adding the lock into my regular home.
Best
Thorsten

thorsten-gehrig, 2024-02-17:
@kormax : can you explain what you mean by "interact with the lock for a bit"?
Is there anything to reproduce this?
With re-adding the lock to the regular home I´m back to the issue with the iPhones - they also always jump to a AUTH1 ... may they dont provision their keys (for whatever reasons).... so i´m curious on digging more into this.
Beside that: is it planed to (re-)implemented the Attestation flow?

Best
Thorsten

kormax, 2024-02-17:
If you read that message again, a couple words before I say that you have to open the Home app on a watch.

thorsten-gehrig, 2024-02-17:
I did that (reading your comment, open the home app on watch, clilcked here and there, opened/closed the lock, ... 
But it didn't worked ... that's why i asked for reproducing this :-)
Anyhow... i´ll give it some more tries ... even I´m not successful yet.
When this happens (provision it's own key into a lock via HAP (network)) - is there something visible on the console-log of the ESP32?

Best
Thorsten

kormax, 2024-02-17:
Yeah,

Sadly, when I did the tests myself, it was really inconsistent. So I cannot guarantee you when or if it will happen. Although it should.

kormax, 2024-02-17:
Actually, if Watch provisioning doesn't happen, I might have a reason for that. 

The protocol has a command that allows to query the configuration state of NFC. 

We were unable to research what it does fully, so we've stubbed some responses.

Perhaps one of them is used to query if a new key has to be provisioned into a lock, and our stubbed response makes it think that no reconfiguration is necessary.

If that's the case, the only short-term solution I see if this project re-adds ATTESTATION support.

thorsten-gehrig, 2024-02-17:
Thanks.
I´ve tried something more intrusive meanwhile: i´ve removed all my hubs (2 Apple TV, 1 AirPod mini) from the home, removed and re-connected the lock.
It does work on 1 phone - and the watch did also provision itself (initially not, then home app, "interact", Boom -> it does work!).
If i´m not mistaken on the console i´ve seen a 
```
I (461042) NFCAccess: update > GET READER KEY REQUEST
I (461042) NFCAccess: update > RESPONSE LENGTH: 16, DATA: xxxxxxxxxxxxx
```
Looks like this was the provisioning process for the keys :-)

That said: somehow the Home-Hubs did cause an issue in my environment!
Hope the re-adding of ATTESTATION support will be done soon :)

Best
Thorsten
(once again: happier today due to progess and success)

kormax, 2024-02-17:
I think the actual provisioning command should be a bit after or before the one you did.

But it's great that it worked for you nevertheless!

rednblkx, 2024-02-17:
isn't this one hell of a thread :))

i haven't seen any issues on my side, though there's not many endpoints in my home(3 iPhones and 2 Watches) and one bridge but i've seen that it might "skip" provision especially when a device is not connected locally.

However regarding the Watch in my tests, they seem to just work, don't need to open the Home app or something though that indeed triggers any "pending things" HAP has to do as i've done this during development.

@thorsten-gehrig could you clarify if you get any other error or warning after "FAST Flow failed!"?

Note that the "FAST Flow failed!" error is normal as there's no persistent key established yet and it will just continue to the STANDARD flow where it attests the key as long as the device credential provision did occur which gives us the public key of the device and if not this is where ATTESTATION flow would come into place to save the day which I hope to (re)add soon, I have been taking a break for the last few days, so no work has been done so far.

rednblkx, 2024-02-24:
@thorsten-gehrig check the new code when you can and let me how it goes, make sure to update the PIO dependencies.

thorsten-gehrig, 2024-02-25:
Hi
the enrolment ist much faster with this code. Unfortunaintly I´ll still have
"*** ERROR: Unrecognized Controller PairingID" once during the enrolment.
I will send logs in a few.
Best
Thorsten

rednblkx, 2024-02-25:
You can ignore that error, is just a device trying to connect with old pairing data. It should refresh itself eventually or you can just open the Home app on your devices so they can sync.

thorsten-gehrig, 2024-02-25:
Ah - ok. I just deleted all home full home to start from scratch (after I´ll removed bridge by bridge...).
Ok - one topic less.
regarding your comment on "update pio dependencies" - I guess "pio lib update" is doing the job, right?

rednblkx, 2024-02-25:
Yes, but note that `pio lib` is deprecated per [docs](https://docs.platformio.org/en/stable/core/userguide/lib/cmd_update.html) and will be removed in the future, instead use `pio pkg` for anything library related, like `pio pkg update` to update.

thorsten-gehrig, 2024-02-25:
Ok-even I´ve seen here and there in my tests a crash (not sure if the dependencies had already been updated) - everything seems working now perfectly!
I´ve re-invited my wife (her access got lost after deleting the old home) - and this works also fine.
I will do some more tests. but i´m very optimistic and will probably soon but it into the production place.

Does OTA update works? I´ve seen some comments - and this will be more important when it´s in production use.

Best
Thorsten

rednblkx, 2024-02-25:
That's wonderful! Glad this sorted it out.
OTA does indeed work, just run the command `pio run -t ota --upload-port <ip-address-esp>`.
I'll be closing this issue now, if there is anything else i can help with, you can drop me an email.

closed 2024-02-25 by rednblkx