rednblkx/HomeKey-ESP32 #26: Lock entity not not getting toggled

ihrapsa, 2024-02-19:
I just updated to the latest main commit and I'm having issues with getting the Lock entity toggled. I can't toggle it manually either.
When using the HomeKey it doesn't react. When doing manually it hangs on "Unlocking" in the Home app with the following logs in serial monitor: 

```
I (179621) LockMechanism: update > New LockState=0, Current LockState=1
I (546199) LockMechanism: update > New LockState=1, Current LockState=1
I (549684) LockMechanism: update > New LockState=0, Current LockState=1
I (552368) LockMechanism: update > New LockState=1, Current LockState=1
I (554962) LockMechanism: update > New LockState=0, Current LockState=1
```
[image]

Is this just me?

rednblkx, 2024-02-20:
Hi Iona, 

The lock is now entirely MQTT-controlled and HomeKey no longer directly interacts with the state instead it's only publishing the authentication data through mqtt on the `MQTT_AUTH_TOPIC` topic and you can decide from there what the state should be with the following control topics:
- `MQTT_SET_STATE_TOPIC`
- `MQTT_SET_TARGET_STATE_TOPIC` 
- `MQTT_SET_CURRENT_STATE_TOPIC`

In case you are not familiar with how the state in HomeKit is handled, know that you have the "target state" which is where you want to go and the "current state" which is where you currently are that way you can have a more fluid flow of changing the state and better indication of success-failure.

The `MQTT_SET_STATE_TOPIC` is controllling both the target and the current state.

When interacting with the lock from the Home app, the target state will be published via mqtt to the `MQTT_STATE_TOPIC` topic. Also, when using HK the toggled state will be published to the same topic which i think i'll remove since it will cause unwanted issues if it's not synced with the state that you want and it would complicates things needlessly to try make it work.

If you are a HASS user, it now integrates with HA's Tags which you can use to create some nice automations.

I'm looking to create a wiki here sometime in the future to detail all the things and perhaps even a discord server if there is the need but right now i'm still focusing on making the code more structured and stable.

ihrapsa, 2024-02-20:
I see, thank you for the thorough explanation and clarification!

Was there a specific reason to why the direct interaction with the HK lock state was dropped? I was hoping to have the lock as standalone as possible for reliability. That way I wouldn’t have to rely on a separate mqtt broker. 

Having mqtt support is really valuable! However, to me, it would make much more sense to have it control the HK lock state independently and then publish mqtt states if you want that synced with HASS or other systems. 🤔

Iona

ihrapsa, 2024-02-21:
I just had some difficulties with the`set_current_state` and `set_target_state` in HASS automations, but luckily that's what it needed to enable the Home app entity toggle on HomeKey scan 🥳 . 

Relay/GPIO control is indeed much anticipated and preferred in my case since that reduces complexity and risks of failure.  Having it battery powered is also on the list but power outages are not that common here.

My use case is with an indoor retrofit BLE battery powered lock that hopefully I'll be able to  tap into it with a relay/optocoupler in order to trigger the lock/unlock sequences. There's also the cleaner alternative of using the BLE API to control it remotely but that would require a bit more code work.

I really appreciate your work on this. 🙂 

Iona

rednblkx, 2024-02-21:
Great, glad you figured it out!

GPIO Control will be added more further down the line as i currently have some other things that i need to take care first.

Talking about battery-powered, that is an entire beast on it's own since you have to account for the whole hardware from the ground up, you can't just use any pre-made board since most development boards have a voltage regulator and usb-ttl converter with a high quiescent current and that's just one issue and when i am going to attempt to make it battery-powered, i will have to switch to BLE as Wifi is not really suited for battery applications.

Since you managed to make the lock work, i'll be closing this issue.
If you have any other issues, you can also contact me on Discord `@rednblkx`

exrector, 2024-06-30:
> Во-первых, я хочу уточнить, что HomeKey не связан с фактической блокировкой сам по себе, на самом деле это отдельная «Сервис» в HomeKit, и именно так HAP разделяет различные объекты общего аксессуара.
> 
> Я в основном удалил "прямое взаимодействие", так как это была просто фиктивная вещь, которую я поставил для тестирования перед введением mqtt, чтобы избежать конфликтов или нежелательного поведения. Однако я собираюсь переработать эту часть, возможно, просто сделать вариант, чтобы она действовала как фиктивный переключатель, как раньше (но все еще имела бы управление MQTT) или, может быть, реализовать какое-то оптимистичное состояние, хотя я думаю, что последнее может оказаться слишком сложным, в любом случае любые предложения приветствуются.
> 
> Примечание: я работаю над веб-интерфейсом в ветке разработки, поэтому любые настройки и параметры будут включены туда в будущем после его объединения.
> 
> Кроме того, хотя я понимаю ваше беспокойство по поводу необходимости полагаться на брокера mqtt, обратите внимание, что в проекте в настоящее время нет реализации GPIO, и поэтому для обновления своего состояния ему приходится полагаться на внешнюю обратную связь.

Good idea to "bring back" the dummy switch!!! Because the mqtt broker is not needed, if you can do everything by simple automation in the home application without unnecessary bridges and brokers. That's what the matter protocol was invented for after all.