rednblkx/HomeKey-ESP32 #37: [FR] Ability to keep lock state after reboot

ihrapsa, 2024-03-11:
I’ve noticed that after **_power outages_** or randomly (probably due to ESP32 **_reboots_**) the state of the lock changes to unlocked in Home app, although that does not trigger any `mqtt` message to be sent in order to mess with automations. However, when in that state (HomeKey lock entity state unlocked but actual lock locked) it messes with the HomeKey authentication and automation logic in the sense that when scanning the HomeKey it will broadcast a `lock` mqtt message when I actually want to `unlock` the door, so nothing happens. If a second HomeKey authentication is attempted, the actual lock will unlock since they are now both (HomeKey entity and actual lock) in sync.

Anyone else has this issue?
Am I doing something wrong? 
Is there any way we can keep the lock state after reboots/power outages?

Thanks!

rednblkx, 2024-03-12:
strange, that is not what i'm seeing on my side, HomeSpan has the option to save a characteristic's value to nvs which i set to true for the lock's target and current state [here](https://github.com/rednblkx/HomeKey-ESP32/blob/e736a9e90a3d58343df7f108f310d3a26317d0b6/src/main.cpp#L115C1-L116C68).

Could you try to change the lock's state (HAP) manually and then reboot the esp and see if the state persist? This is to understand if somehow there is something else going on.

On a side note, also as a workaround, if you don't have the esp hardwired to a lock, don't rely on the HAP state for your automations since it is kinda arbitrary, instead make the decision to unlock or lock around your real lock's state based on the MQTT_AUTH_TOPIC topic or even better if you use HASS you can use Tags since the project has integration through mqtt discovery for that and you can very easily make automations at a click of a button per person/device.

ihrapsa, 2024-03-12:
Thanks, will do a test to check if HomeSpan saves the state. 

Regarding the `MQTT_AUTH_TOPIC` what happens if a HomeKey from a different home is presented?  Or in other words is the HomeKey mqtt message only transmitted if HomeKey authentication succeds? I’m wondering if I could just use the `homekey=true` as a trigger for unlock.

Jonah

rednblkx, 2024-03-13:
Well…yeah, I don’t think it would make sense to publish on all attempts, just for the successful ones. If you have multiple deployed and are all using the same topics, it should be fine, the endpointId is unique per setup as each device generates a key for each lock, only the issuerId stays the same.

The `homekey` field exists only to indicate the message contains homekey data since regular rfid cards would publish different data, although you could use it as a basic validation that the data is homekey related kinda like this:
```yaml
## Basic HASS Automation
alias: Homekey
description: ""
trigger:
  - platform: mqtt
    topic: topic/auth
condition:
  - condition: template
    value_template: "{{trigger.payload_json[\"homekey\"] == true}}"
action:
  - if:
      - condition: device
        device_id: 
        domain: lock
        entity_id:
        type: is_locked
    then:
      - device_id:
        domain: lock
        entity_id:
        type: unlock
    else:
      - device_id:
        domain: lock
        entity_id:
        type: lock
mode: restart

```

ihrapsa, 2024-03-14:
I did some further tests and the Home entity keeps state on reboot when manually triggering it over `mqtt`. I'm still not sure what causes that, but I've decided to avoid using the dummy entity in any automations for the time being. It's just annoying to see it in the Control Center as `Unlocked` in the middle of the night when it actually is `Locked` 😳. 
We can close this one.
Thanks a lot!

rednblkx, 2024-03-15:
The functionality was added about 2 months ago so i doubt that you have an old version. I did multiple tests uncoupled from any automations and it is able to persist state on power outages(a.k.a unplugging it) and also checked the activity history up to one month and there's no random state change.  so really not sure why's that way for you, sorry about that. I'll close the issue for now since i'm not able to reproduce it but if something comes up i'll reopen it.

closed 2024-03-15 by rednblkx