rednblkx/HomeKey-ESP32 #42: No recovery from failed PN532 communication

pnuding, 2024-03-24:
I’ve got a reader in a place with a wifi antenna, switching power supply and 230v wiring all nearby. 
It was rock solid when I tried it for a week at my desk without these noisy neighbours. 
But with all this interference it seems to loose the communication with the PN532 after a while and when it does it never recovers until i reset it. 
In this state it’s reachable via Mqtt and works normally in HomeKit but no longer responds to nfc tags or homekeys. 
I’ve managed to improve the wiring SPI placement so it now lasts a day instead of a few hours, but it would probably be a good idea to retry if SPI with the pn532 fails

rednblkx, 2024-03-26:
Hi there!

Thank you for reporting this, it is on my list of things to improve, however, the issue is, unlike I2C there is no real way of detecting the presence of an SPI slave but there is some code trickery which can be done that i'll be looking into. 

Side Note: It was actually supposed to be I2C but i had some issues with it so i picked SPI instead, i might actually try to make I2C work to see how it behaves.

pnuding, 2024-03-26:
Perhaps it's as simple as starting over with the SPI communication every now and then. That alone would make it much more robust irrespective of whether it's necessary in that moment or not

da2001, 2024-05-26:
sadly i ran into the same problem. worked perfect on my desk but no it works for just some hours and than needs a "reboot".

pnuding, 2024-05-26:
I've managed to solve it on the hardware side here - lots of aluminium tape and moved the esp32 to a box apart from the other stuff (and some aluminium tape around the walls of the box, just in case 😆 )

closed 2026-01-29 by rednblkx