rednblkx/HomeKey-ESP32 #48: Running on ethernet (WT32-ETH01) -  restart loop every 10 sec

DP365, 2024-04-13:
I have an WT32-ETH01 (ESP32 with LAN 8720 ethernet chipset and RJ45 port) and HomeKey-ESP32 works fine in WiFi mode on this board (needed to change pins_arduino.h to make it work: SS = 5; MOSI = 23; MISO = 19; SCK = 18;). I was then adapting the Ethernet modification for HomeSpan (https://github.com/HomeSpan/HomeSpan/discussions/738) and was trying to integrate it with HomeKey-ESP32. The program launches and ethernet connection works, the WT32-ETH01 reacts to random NFC cards and I can see MQTT messages coming through. But every 10 sec the device does a restart with the message "*** FACTORY RESET!  Restarting...".

rednblkx, 2024-04-13:
Are you by any chance using the Pin 26? That's assigned as the control pin by default in the config.h file.

Regardless, according to this [comment](https://github.com/HomeSpan/HomeSpan/discussions/738#discussioncomment-8830561) in the same discussion, Ethernet support is supposedly coming in the next HomeSpan release so i think it's best to just wait for that.

DP365, 2024-04-13:
Ah that was it, Pin 26 is hardwired to the LAN8720 so I moved the CONTROL_PIN to 5 and now it works on ethernet as well. Great project btw!!!

rednblkx, 2024-04-13:
Isn’t 5 assigned to SS?

DP365, 2024-04-13:
SS was on 4 - I edited my first post accordingly

rednblkx, 2024-04-13:
Ok, great, I’ll be closing this then.

closed 2024-04-13 by rednblkx

hallerdavid, 2024-07-23:
Hey, thank you for the project and thank you @DP365 for testing this with an WT32 ETH01.

Can you help me? I have now connected the WT32 ETH01 with an PoE Splitter, connected 5V and GND to the WT32 ETH01 AND to GND and VCC from the NFC reader board.

Can you help me what numbers you mean with 4, 23, ... my board only have other texts on it... :-D maybe i´m a little bit dump. :-)

DP365, 2024-07-23:
hey @hallerdavid - apologies, there were some more typos, I updated my initial post.

hallerdavid, 2024-07-24:
thank you for the complete path, i´m a totally newbie so what pins from the NFC board should i soldier to what pins from the ESP board? maybe you have some pictures :-D

DP365, 2024-07-24:
Check the wiki but use the pins above on the WT32-ETH01 side -- https://github.com/rednblkx/HomeKey-ESP32/wiki/Prerequisites

drstil, 2024-08-15:
@DP365 Thank you very much for sharing this information! 

> I was then adapting the Ethernet modification for HomeSpan (https://github.com/HomeSpan/HomeSpan/discussions/738) and was trying to integrate it with HomeKey-ESP32

Did you only copy the mentioned code to `main.cpp`? And then compile as mentioned in [old setup](https://raw.githubusercontent.com/wiki/rednblkx/HomeKey-ESP32/Old%20Setup.md)?

@rednblkx 
> Ethernet support is supposedly coming in the next HomeSpan release so i think it's best to just wait for that.

Seems that ethernet is not supported by HomeSpan till now: https://github.com/HomeSpan/HomeSpan/releases
>HomeSpan v1.9.1: for future integration of Ethernet support

Do you have further information when ethernet support will be release by HomeSpan?

rednblkx, 2024-08-16:
> Do you have further information when ethernet support will be release by HomeSpan?

Unfortunately I have the same information, I might be doing my own implementation for it since I’m already using a fork of HomeSpan, however, I’m working towards moving away from the Arduino framework and inherently HomeSpan.

drstil, 2024-08-20:
Here is my running version for WT32-ETH01. Flashed by PlatformIO as described in [old setup](https://raw.githubusercontent.com/wiki/rednblkx/HomeKey-ESP32/Old%20Setup.md).
[main.cpp.zip](https://github.com/user-attachments/files/16678777/main.cpp.zip)