rednblkx/HomeKey-ESP32 #54: NFC Reader

s31teg, 2024-04-24:
Hello

I've got a spare one of these RFID readers. Do you know if this will work with the esp?

https://thepihut.com/products/rfid-breakout-board#product-reviews

TIA!

011V32, 2024-04-24:
Is this a general question, or should this be related to this project?

ihrapsa, 2024-04-24:
> Hello
> 
> I've got a spare one of these RFID readers. Do you know if this will work with the esp?
> 
> https://thepihut.com/products/rfid-breakout-board#product-reviews
> 
> TIA!

I assume you want to use this project for it. In that case it won't work. You need a PN532 based NFC reader which is 13.5h MHz. Your board is a 125kHz RFID reader.

s31teg, 2024-04-24:
Thanks for the clarification. 

On another hardware question have you seen a poe hardware esp32 board you would recommend?

Would something like this work and the code compile on?
https://www.mouser.co.uk/ProductDetail/Olimex-Ltd/ESP32-POE-ISO?qs=unwgFEO1A6vVfCRbLeeCIw%3D%3D

s31teg, 2024-04-24:
Ahh ok, fortunately, it seems will be soon :)

I currently use a loxone system to control my FUHR multitronic 881. Looking for the neatest and most responsive way to integrate homekey into it. Thanks for your code anyway I will test the performance of the solution and hopefully a hardwired option will come that I can integrate directly without MQTT

ihrapsa, 2024-04-24:
All credits go to @rednblkx not me. Good luck!

rednblkx, 2024-04-26:
Hi @s31teg 

Appreciate the interest on this project!

Don't have much knowledge about PoE but i know it is a hardware-level functionality so it will work regardless of the software, Ethernet on the other hand, is what is not yet implemented in HomeSpan so you would still need to use WiFi, at least for the moment.

The code should compile and run just fine on the aforementioned board and a "hardwired option" is in the works and will come out along with the web interface as that will make the most sense, no ETA on that unfortunately.

closed 2024-05-08 by rednblkx