rednblkx/HomeKey-ESP32 #55: Didn't find PN53x board

budisek, 2024-04-25:
[image]

I have this error. PN532 is correctly connected with LED on. I can't even pair lock with Apple HomeKit.

What I'm doing wrong?

rednblkx, 2024-04-26:
Hi there!

I've previously had a similar report, so we'll start there. 

How are you powering the PN532? Hopefully from the board as noted in the guide as they have to share a voltage reference.

Are you using the 5V or the 3V3 pin?

budisek, 2024-04-27:
Powering is from the board. I tried 3v3 and 5v. LED od the RFID board is on but it doesn't do anything.

topt, 2024-04-28:
I have a slightly different ESP32 board, but same issue - NFC reader not recognized:

```
I (2997) gpio: GPIO[5]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3000) gpio: GPIO[18]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3009) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3019) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (3041) NFC_SETUP: Didn't find PN53x board
```
I have Wifi working, I get MQTT updates in HomeAssistant, I can flip the Lock using the Home app (and get the updates in HomeAssistant), but the NFC reader does not recognize anything (no iPhone, also not Mifare/ISO14443 cards or NTAG213 or similar.
the NFC reader is connected to the VCC (5V) of the ESP, GND aswell. Model is a red PN532 from elechouse, but may be one of the 'cheap' variants. I double checked the cabling, should be as documented. Any ideas on how to debug?
I also tried another Poweradaptor (besides the USB connection to the pc).

budisek, 2024-04-29:
I have no idea what is wrong.

IvanDeMarco, 2024-04-30:
I have the same problem, any ideas?

gpambrozio, 2024-05-01:
Ran into the same issue. My board has this really small dip switch and I had to flip switch number 2 so the board would work over SPI. This is how I have it working now.

[image]

budisek, 2024-05-01:
@gpambrozio You are right. This solved my issue. It should be mentioned in Setup Wiki

VaBanck, 2024-05-01:
> gpambrozio You are right. This solved my issue. It should be mentioned in Setup Wiki

A PN532 NFC Module
This needs to use the SPI protocol
For the red board from Elechouse, there should be a DIP Switch on it and a small table to the side that indicates the switches states for each protocol, for SPI this should be 0 and 1 which means left ⬅️ switch down ⬇️ and right ➡️ switch up ⬆️

IvanDeMarco, 2024-05-01:
I had already moved the DIP switches for SPI operation but it still doesn't work…

011V32, 2024-05-01:
> @gpambrozio You are right. This solved my issue. It should be mentioned in Setup Wiki
> 
> 

https://github.com/rednblkx/HomeKey-ESP32/wiki/Prerequisites#hardware-and-software

rednblkx, 2024-05-08:
I've been working on refactoring the pn532 driver to use the esp-idf APIs and there seems to be an issue on the driver with SPI and the waking up sequence of the pn532 though that should only affect the first command but in some cases it starts a chain reaction, however, i've managed to find a workaround(not sure if it's also the solution).

Anyone that still has issues, please try the `esp-homekit-sdk` branch and let me know if it still shows the same error.
Note that this is not a usable version at the moment, it lacks the mqtt implementation, it's just a environment to test the nfc driver.

topt, 2024-05-12:
Hi @rednblkx, I tried your `esp-homekit-sdk` branch, unfortunately same result for me, does not find the board.
To rule out a general issue with the board, I gave it a try with the plain esphome pn532 library. Using SPI same issues, I cannot get the board to work. I then tried I2c (changing the DIP Switches), and this works on ESPHome (just basic reading of Mifare cards). So it look like the board itself works, but I found others having similar issues in general with this elechouse V3 clone using SPI. So currently I am unsure if this is a weakness of this board in general, or if mine maybe has broken SPI support. Would your project also work on I2c in general? Or do you see any ways forward in maybe additional debugging I could help to get it to work? Thanks.

closed 2024-05-16 by rednblkx