rednblkx/HomeKey-ESP32 #56: Home key not working on home kit

moshiko2312, 2024-04-27:
Hi thanks for your job
Ido all in the document and all running well.
The mqtt in home assistant working and talking with homkit well.
But if im scanning my iphone its give me OK but nothing do in home kit.
What i need to do to fix that?
Thanks for all
[image]
[image]

rednblkx, 2024-05-08:
HI there!

Sorry for the late reply.

The Homekey and HomeKit states are not coupled together, that lies on the lock's responsibility to update the state and in the case of this project it is done through MQTT (GPIO Control is on my list to add) which results in the state remaining unchanged if no update is sent.

Let me know if you have any other issues or questions.

moshiko2312, 2024-05-09:
Hi thanks
Its working im suing shortcut from iphone to do the logic and get the
trigger from the NFC reader.
And actually control directly on Nuki.

בתאריך יום ד׳,

rednblkx, 2024-05-09:
Glad you got it working!

Know that for Nuki you can also use the lock's MQTT API and project's MQTT Custom states

See Issue #38 and the [wiki](https://github.com/rednblkx/HomeKey-ESP32/wiki/Usage#111-mqtt-custom-states)

Let me know if you have any questions.

moshiko2312, 2024-05-09:
Hi im using nuki mqtt directly to HA
But all the logic im do from iphone shortcuts.
But it's not perfect because need internet to send the trigger to the
system.

Its gonna be good if your NFC code can open the Hlock directly.
So it's be local and no need to use iphone shortcuts.

בתאריך יום ה׳,

rednblkx, 2024-05-10:
Like i said, you can use this with Nuki's MQTT API, just enable `MQTT_CUSTOM_STATE_ENABLED` and assign `MQTT_CUSTOM_STATE_TOPIC` to Nuki's command/control topic and `MQTT_CUSTOM_STATE_CTRL_TOPIC` to Nuki's state topic.

Or since you mentioned using HA, i don't see why you are not using HA to handle the logic?

moshiko2312, 2024-05-10:
Hi because I don't understand how to triggering fron the NFC reader to the
system whiteout iphone shortcuts.
Can you show me or let me know how to do that?
Thanks

בתאריך יום ו׳,

rednblkx, 2024-05-10:
Sure, the easiest way would be to just use Home Assistant's Tags as the project has integration with that. 

If you haven't used Tags before, in HA's Settings there is a Tags section and there you should see a list of tags, should be at least two, as HomeKey uses two for identification, one for each of your devices (endpoint id) and another one identifying your Apple ID (issuer id) so you can have one ID that you can use to authorize your devices individually and another one that just authorizes you.

You can easily distinguish them by clicking on the cog icon(still in the Tags section) and looking at the Tag ID, the shorter one is for your device, also when scanning your key you should see a ripple effect on the tag that got triggered, that way you can identify between multiple issuer and endpoint ids which belongs to what.

rednblkx, 2024-05-10:
You then just need to click on the robot icon for each tag that you wish to have as a trigger in order to create an automation

moshiko2312, 2024-05-10:
> Alternatively, you can also use this simple automation
> image.png (view on web)

Thanks man.
I will do that and let you know
Thanks fir your job🙏

בתאריך יום ו׳,

moshiko2312, 2024-05-11:
Ok
Using now with Tags its working perfectly and locally.
Thanks for all.🙏

בתאריך יום ו׳, 10 במאי 2024 ב-21:54 מאת moshiko peer ***@***.***

rednblkx, 2024-05-11:
Great, glad to help! 🙏

Let me know if you have any issues.

closed 2024-05-11 by rednblkx