rednblkx/HomeKey-ESP32 #57: buzzer and LED

Epik77, 2024-04-30:
HI,
I would like to implement your project on my home automation.
I saw how your device works and with a small plugin that I wrote in Python I am able to integrate it into my domoticz hub
I wanted to ask you if you could include in the code the use of a simple buzzer (for reading confirmation of the NFC device) and some LEDs to be used on MQTT to be turned on for your own use (you can integrate it with DISCOVERY of Home Assistant)
Thank you very much for your time on this project.

rednblkx, 2024-05-08:
Thank you for the interest in this project, appreciate it. Apologize for the late reply, been a busy week.

The buzzer and LED are on my to-do list and i think i'll be getting to it sometime soon.

I'll let you know here when that happens.

rednblkx, 2024-06-16:
See release notes for the latest version
https://github.com/rednblkx/HomeKey-ESP32/releases/tag/v0.3

Right now MQTT exposed LEDs are out of scope for this project, but will be revisited on a later version.

closed 2024-06-16 by rednblkx

leschNUX, 2024-06-18:
Hello rednblkx,

Would it be possible to add support for a pwm buzzer like this one

https://www.lyonscomputer.com.au/Electronic-Modules/Pizo-Buzzer-MH-FMD-Module/Pizo-Buzzer-MH-FMD-Module.html