rednblkx/HomeKey-ESP32 #59: Homekit lock not toggle

dorbenzvi, 2024-05-24:
I'm using homebrige and added the HK lock generaed by this project.
Installed Mosquitto on my respberry pi so I'm connected and and looks but.
I also set MQTT_HOMEKEY_ALWAYS_UNLOCK = 1 since I want it only to onlock and I will create an automation to lock it by sensor state.
But I can only unlock it in homekit.
If I click lock it seems like it stuck on loading.
Anyone knows why?

My config:

// MQTT Broker Settings
#define MQTT_HOST "MY_IP"   // IP adress of mqtt broker
#define MQTT_PORT 1884               // Port of mqtt broker
#define MQTT_CLIENTID "homekey_mqtt" // client-id to connect to mqtt broker
#define MQTT_USERNAME "MY_PASSWORD"     // username to connect to mqtt broker
#define MQTT_PASSWORD "MY_PASSWORD"     // password to connect to mqtt broker

// MQTT Flags
#define MQTT_CUSTOM_STATE_ENABLED 0  // Flag to enable the use of custom states and relevant MQTT Topics
#define MQTT_HOMEKEY_ALWAYS_UNLOCK 1 // Flag indicating if a successful Homekey authentication should always set and publish the unlock state
#define MQTT_HOMEKEY_ALWAYS_LOCK 0   // Flag indicating if a successful Homekey authentication should always set and publish the lock state

// MQTT Topics
#define MQTT_CUSTOM_STATE_TOPIC "topic/custom_state"           // MQTT Topic for publishing custom lock state
#define MQTT_CUSTOM_STATE_CTRL_TOPIC "topic/set_custom_state"  // MQTT Control Topic with custom lock state
#define MQTT_AUTH_TOPIC "topic/auth"                           // MQTT Topic for publishing HomeKey authentication data or RFID UID
#define MQTT_SET_STATE_TOPIC "topic/set_state"                 // MQTT Control Topic for the HomeKit lock state (current and target)
#define MQTT_SET_TARGET_STATE_TOPIC "topic/set_target_state"   // MQTT Control Topic for the HomeKit lock target state
#define MQTT_SET_CURRENT_STATE_TOPIC "topic/set_current_state" // MQTT Control Topic for the HomeKit lock current state
#define MQTT_STATE_TOPIC "topic/state"                         // MQTT Topic for publishing the HomeKit lock target state

// Miscellaneous
#define HK_CODE "46637726"     // code used for homekit setup
#define LED_PIN 2              // status led pin, built in is pin 2
#define OTA_PWD "homespan-ota" // custom password for ota
#define NAME "HK Lock"         // Device name
#define DISCOVERY 0            // Enable or disable discovery for home assistant tags functionality, set to 1 to enable.
#define CONTROL_PIN 26         // GPIO Pin for a Configuration Mode button (more info on https://github.com/HomeSpan/HomeSpan/blob/master/docs/UserGuide.md#device-configuration-mode)

rednblkx, 2024-05-26:
Hi there!

When interacting with the lock in the Home App it will publish the requested state via MQTT to the topic `MQTT_STATE_TOPIC` and topic `MQTT_CUSTOM_STATE_TOPIC` (if enabled) and in order for entity to go into that state it has to be set then via the MQTT topic `MQTT_SET_STATE_TOPIC` or one of the other control topics as part of your automation.

In order to debug, use [MQTT Explorer](https://github.com/thomasnordquist/MQTT-Explorer/releases) or something similar to listen to the `MQTT_STATE_TOPIC` and see how the payload changes, when unlocking you should see `4` and for locking it should be `5` as indicated in the [wiki](https://github.com/rednblkx/HomeKey-ESP32/wiki/Usage#11-handling-lock-state)

Let me know how it goes

EliCohen, 2024-05-31:
> I'm using homebrige and added the HK lock generaed by this project. Installed Mosquitto on my respberry pi so I'm connected and and looks but. I also set MQTT_HOMEKEY_ALWAYS_UNLOCK = 1 since I want [...]

@dorbenzvi did you get managed? I have the same issue.

closed 2024-06-09 by rednblkx