rednblkx/HomeKey-ESP32 #60: Esp32 WT32-ETH01

moshiko2312, 2024-05-31:
Hi how its possible to working with ESP32 WT32-ETH01
Its esp with LAN Cable.
I don't want working on WiFi
Thanks for all your job its the best way i see🙏

Just found this amazing item on AliExpress. Check it out! ₪37.61  67％ Off | WT32-ETH01 Embedded Serial Port Networking Ethernet Bluetooth-compatible WIFI Combo Gateway MCU ESP32 Wireless Module Board
https://a.aliexpress.com/_EvBJprt

011V32, 2024-05-31:
> Hi how its possible to working with ESP32 WT32-ETH01
>
> Its esp with LAN Cable.
>
> I don't want working on WiFi
>
> Thanks for all your job its the best way i see🙏
>
>
>
> Just found this amazing item on [...]

At the momement no LAN Support. But will come hopefully in near future. See other posts.

moshiko2312, 2024-05-31:
Ok thanks
Its working perfectly

בתאריך יום ו׳,

moshiko2312, 2024-06-29:
I buy this one

https://shop.everythingsmart.io/products/est-poe-32-esp32-poe-development-board

magic2910, 2024-07-07:
Is the board from aliexpress working?

ChrisKoh83, 2024-07-23:
@moshiko2312 
To get this working I guess you had to build your own code to make it work?

because the GPIOs used by the Ethernet module!?

DP365, 2024-07-23:
yep, correct. you need to use the second hardware SPI bus (called HSPI) because the first one (VSPI) is used by the ethernet module. See also my post here https://github.com/rednblkx/HomeKey-ESP32/issues/48#issue-2241602306

klim123123, 2024-08-13:
Its just works for me. Nothing special, i used random GPIOs.

```
spi:
  clk_pin: GPIO14
  miso_pin: GPIO4
  mosi_pin: GPIO2

pn532_spi:
  id: nfc_spi_module
  cs_pin: 12
  update_interval: 100ms
```

drstil, 2024-08-15:
> Its just works for me. Nothing special, i used random GPIOs.
> 
> ```
> spi:
>   clk_pin: GPIO14
>   miso_pin: GPIO4
>   mosi_pin: GPIO2
> 
> pn532_spi:
>   id: nfc_spi_module
>   cs_pin: 12
>   update_interval: 100ms
> ```

Thank you for this reply! How did you set those settings? `config.h`? How did you compile afterwards? Thank you very much!

klim123123, 2024-08-15:
> @klim123123 Thank you for this reply! How did you set those settings? `config.h`? How did you compile afterwards? Thank you very much!

Sorry, i'm confused with other project of rednblkx.
I'm using rednblkx/HAP-ESPHome
Same functionality but for ESPHome. It's just works.