rednblkx/HomeKey-ESP32 #62: Can't get Wifi connected

x23piracy, 2024-06-06:
Hi,

i compiled and flashed on a ESP32-WROOM-32, it's booting but there is constantly output flowing over the screen i cannot press W + Enter or A nor it's doesn't react to the cables shortended between GND and GPIO26, what i am doing wrong?

I can also upload pictures and or console output if i am able to capture because its constaly repeating things.

Best Regards X23

011V32, 2024-06-07:
You need to configure your serial terminal to the right line ending. (I think it is /n only, but look in the documentation of Homespan.)

x23piracy, 2024-06-07:
Hi,

yes you saw it used 115000 instead of 115200 anway this changes nothing the output is still scrolling and i looks like its resetting all the time, it doesnt react to my input.

Best Regards X23

x23piracy, 2024-06-07:
Sure here is a picture, got 3 and i already used one of them for ahoy dtu.

[image]

011V32, 2024-06-07:
The reboot not, but that it didn't react to the Input could be the line ending. I had the same problem...
The Terminal on windows needs to have LF also for transmitting (Putty only over Ctrl+J instead Enter, so I choosed Tera Term instead)

rednblkx, 2024-06-07:
@x23piracy the RC522 is not supported, you need a PN532 for it to work. Also note that even though esp32 is low power compared to rpi it will still use some amount of power that it won't be suitable to be powered by a battery long-term if you were planning on doing that as it doesn't sleep due to HomeKit needing a persistent connection.

rednblkx, 2024-06-07:
> The reboot not, but that it didn't react to the Input could be the line ending. I had the same problem... The Terminal on windows needs to have LF also for transmitting (Putty only over Ctrl+J instead Enter, so I choosed Tera Term instead)

That yes though what you are referring is what windows uses as a standard which is actually `CRLF` not `LF`, that is actually used by Unix systems which does not matter for Serial as that's set by the device not on host which in this cause is `CRLF` for receive and `LF` for transmit which are esp-idf defaults.

x23piracy, 2024-06-07:
Hi,

i just told ya i have currently a RC522 in place with a wemos, the ESP32 is complete other thing i have everything here so all cool 🗡️

rednblkx, 2024-06-07:
Glad it works now, though very strange. 

Note that the key and the lock are not coupled together, they are two separate things and by default an authentication event does not lock or unlock the "HK Lock" directly as denoted in the wiki.

This has slightly changed in the dev branch along implementing support for gpio status and action pins, more will be detailed in the wiki once released.

for the error above with the dev branch, it's because i put `/` in the spiffs image location as i use Linux and Windows uses `\` as a path separator, will be fixed once merged into main. The dev branch now implements a webui which has the html files stored in the spiffs partition and it is flashed separately hence the spiffs image.

In the meantime if you still want to try it, you can open the `fs.py` and just replace the `/` in  `$BUILD_DIR/$ESP32_FS_IMAGE_NAME` or delete the following line from `platformio.ini` to stop the script from running since we are only testing to see if it "boots"
`extra_scripts = pre:extra_script.py, pre:fs.py`

x23piracy, 2024-06-07:
i just played around with that i have now at you need to really come close to the 532, the card and tag have a much better distance, detected around 3-4cm away from the board, i think this is by design (apple)?

x23piracy, 2024-06-07:
It would be really cool if this in the end will have a webif where we can add tags that are allowed, i did this with tasmota and the RC522 combined with rules, i wrote a salt into block01 to not just have the uid which maybe can be easily bruteforced (4 times a second) therefore i created rules within tasmota which check for uid and my salt if both match i open the lock which is currently made with homebridhe and http lock plugin :) a lot of combining this and that :D

rednblkx, 2024-06-07:
> i just played around with what i have now at you need to really come close to the 532, the card and tag which have been shipped with the 532 have a much better distance, detected around 3-4cm away from the board, i think this is by design (apple)?

yes and no, you can actually get 5-6cm if you have a reader with a proper antenna, that's why it's best to get the original module from Elechouse if possible, though, i was looking at ways to shield from EMI and place the reader on metal surface and discovered you can place piece of ferrite sheet and aluminium foil on the back of the reader and it focuses the field better and as a result increases the range a bit to like 5-6 cm or at least it worked that way with one of the modules i bought.

Also if you have an apple watch, you might notice that it has a better reception and that's because compared to an iphone which has a small antenna in the left corner, the watch has a bigger antenna the size of screen

> It would be really cool if this in the end will have a webif where we can add tags that are allowed, i did this with tasmota and the RC522 combined with rules, i wrote a salt into block01 to not just have the uid which maybe can be easily bruteforced (4 times a second) therefore i created rules within tasmota which check for uid and my salt if both match i open the lock which is currently made with homebridge and http lock plugin :) a lot of combining this and that :D

that's really cool, i do plan to add whitelisting tags and possibly some other stuff but i also want to make it as stable as possible and also more secure.

rednblkx, 2024-06-07:
i am just now introducing a basic webui once i merge dev so there is a long way

x23piracy, 2024-06-07:
Cool, i stay tuned and will look for this project :) no hurry i have a solution in place but like the idea to use home key, so what i found is that you need to authenticate against the tag depending on the technology (mifare etc.) then you are able to write data to block 01, is the PN532 able to write or can it just read?

rednblkx, 2024-06-07:
> is the PN532 able to write or can it just read?

the module itself can of course do both, however, this project currently only reads the uid the tag and publishes it via MQTT.

> what i found is that you need to authenticate against the tag depending on the technology (mifare etc.) then you are able to write data to block 01

yes, cards like mifare classic for example, have a series of keys for each their sectors(which contain the blocks) and without keys you can't read the sectors. At some point i am thinking of adding stuff like that so you can have options but for the moment trying to figure out the structure of the whole thing.

> hmm the home key doenst seem to appear on the old apple watch, meh series 4 and up. I get me a new one :D

well yeah, homekey is relatively new, ios 15

x23piracy, 2024-06-07:
can also publish data thats what i wrote to block 01

x23piracy, 2024-06-07:
what is the real front of the eletech module? the site with the led where the antenna is printed on the pcb? if so it maybe makes sense to solder the cable directly from the backside, maybe the connector interference the communication, i recieve really often an error try again.

How exactly place a shield of metal? do you have a picture?

x23piracy, 2024-06-07:
just tried apple watch se from my wife, not successful just states try again. I had 4 attempts, now shes gone with her watch :D

rednblkx, 2024-06-07:
> just tried apple watch se from my wife, not successful just states try again. I had 4 attempts, now shes gone with her watch :D

trial over, now buy your own :)))
joke aside, i think it might have failed provisioning so it's missing ids and keys. Try the following:
- Remove the accesory
- use command `@LD` in the serial terminal to enable debug log
- Add the accessory back in Home
If you are using PIO's console, output should be saved to file in the logs folder, send me the content from the moment you added the accesory, just make sure to remove parts from the actual hex data to not feed private data into the public web and i don't need it, you can also email me if you want.

> how did you isolate against shorts? looks like double sided foam tape, isn't it?
It's not, i told you, it's ferrite sheet(actually magnetic sheet for photos), stacked on top of each other, the ferrite acts as an electromagnetic barrier to shield the antenna from interference and combined with the aluminium foil blocks the rf from the block and just bounces back i guess :shrug: , don't know exactly the whole process for the aluminium foil.

> am i able to change the gpios? so i can maybe solder a solid 4 pin slot unter the 532 so i can connect everything expect vcc and ground by a stick connection. what a pitty no way to assemble both without the use of cable.

power doesn't flow through the spi pins so you have to power the nfc module through its vcc and gnd pins

> Maybe its better to keep the esp32 away from the reader, maybe interferences or so?

i think generally it's better to keep them away from each other, however, i still just stacked mine 😅 as i wanted to print a small square case.

x23piracy, 2024-06-08:
I tried aluminum/thin double sided tape/paper, decreases distance heavy.

[image]

rednblkx, 2024-06-08:
not sure how you did that but i'm glad it worked out for you, that's hilarious 😂

x23piracy, 2024-06-08:
It seems to be the position or earth radiation idk :D i try to make a video of it.

rednblkx, 2024-06-08:
the issue is that per your description, you don't have a ferrite layer between the module and the aluminium foil. If you only have the aluminum foil, it absorbs the magnetic field, but idk maybe there is something in your double sided tape 😂

rednblkx, 2024-06-08:
not an expert however, just stuff i've found during my research and what i've observed during my tinkering

x23piracy, 2024-06-08:
what do you  mean with control and state topic, state is match but control? 

[image]

rednblkx, 2024-06-08:
> what do you mean with control and state topic, state is a match but control?

the control topic is where you send your command to the lock and the state topic is where you receive the current state of the lock

based on Nuki's MQTT API documentation, the state topic should be `nuki/hex_id/state` and the control topic should be `nuki/hex_id/lockAction`

So then the configuration would be like this:
 - `MQTT_CUSTOM_STATE_ENABLED 1`
 - `MQTT_CUSTOM_STATE_TOPIC nuki/hex_id/lockAction`
 - `MQTT_CUSTOM_STATE_CTRL_TOPIC nuki/hex_id/state`

> i can see without enabling anything it's quiet silent here:

`homekey_mqtt` is an "internal topic" and only contains `status` as it only serves as LWT and availability topic for HASS as it requires that.

You need to look into the topics you have defined in the "MQTT Topics" section

x23piracy, 2024-06-08:
Hi thanks for helping here out, that works well i also enabled MQTT_HOMEKEY_ALWAYS_UNLOCK 1 so it will just always open the lock but doesnt pull the trap how can i archive another lock state when unlocking or in the event it is already open and need to open the trap.

its lockAction 2 and 3

rednblkx, 2024-06-09:
> Is there a way to disable the led on the PN532 board and maybe use one of the gpio pins on the esp32 for an alternative led? is it maybe also possible to connect a beeper for an acustic feedback?

The LED on the PN532 board is hardwired so you'll have to desolder it.
Regarding assigning a gpio pin on the esp32 for an led or a buzzer, that is one of the things that is already implemented in the dev branch

> Wishlist:
> 
>     * OTA Update
> 
>     * kept settings
> 
>     * webif with all that incredible options to change :P
> 
>     * RFID/NFC dongle support (Whitelist), maybe add Security (Block 01)

1. OTA updates already present as part of HomeSpan, keep forgetting to add entry about it on the wiki, i only did a short paragraph on the README
2. Not sure i understand, could please give me more details on this?
3. A basic web interface has already been implemented in the dev branch with all the current options from `config.h` and a little more
4. Tag whitelisting is on the map but i don't currently plan on implementing this very soon

Once the dev branch is ready, i'll have updated the wiki pages and be merged into main with an entry in the releases page detailing all that has chanegd.

> Lol so much wishes, whats your paypal :D
> My issue is solved and can be marked as those, is there another way to communicate with you?

I don't use paypal but you can choose to sponsor me on github if you wish to do so and can contact me by email on rednblkx@protonmail.com or on Discord `@rednblkx`

Since the original problem has been fixed(or rather fixed itself, which is always nice 🫠), i'll be closing this issue now, if you have a subject to talk about, a question or a suggestion, you can also create a new discussion in the discussions section.

closed 2024-06-09 by rednblkx

x23piracy, 2024-06-09:
For 2. i ment making the options changeable via webif persistently. I'll try dev again with the backup devices i have, i can then also test ota with it before i always need to disassemble the front door setting.

x23piracy, 2024-06-09:
i tried current dev but after flashing i get:

`[22:11:56]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:11:56]configsip: 0, SPIWP:0xee
[22:11:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:11:56]mode:DIO, clock div:2
[22:11:56]load:0x3fff0018,len:4
[22:11:56]load:0x3fff001c,len:1044
[22:11:56]load:0x40078000,len:8896
[22:11:56]load:0x40080400,len:5828
[22:11:56]entry 0x400806ac
[22:11:57]ets Jul 29 2019 12:21:46`

rednblkx, 2024-06-09:
Looks like it didn’t flash right and it’s not able to boot the firmware so it just keeps resetting.

I know you’ve had an error previously with the spiffs, how did you handle it?

x23piracy, 2024-06-09:
I used ESPHome Flasher for the bin, i should try with vcode but i still get an error, i replaced / against \

[image]

[image]

rednblkx, 2024-06-09:
well there's the issue, you're missing partitions, the bootloader, partitions header and ota_data

The thing with the ESP32 toolchain is that once the code is compiled, you get 4 .bin files(or more depending on partitions configuration), bootloader.bin, firmware.bin where the actual code lives, partitions.bin which defines the structure and an ota_data_initial.bin which stores data for the current running partition.

The tool that builts littlefs.bin might be missing on windows or something, this only stores the html files for the webui, i'll take a look when i can to see what's going on with Windows, in the meantime if you want you can flash the image i've attached below, i've merged the images into one, just use `[esptool.py](https://github.com/espressif/esptool/releases/download/v4.7.0/esptool-v4.7.0-win64.zip)  like so `esptool.py write_flash 0x0 hk_dev.img` and that should get you up and running.

[hk_dev.bin.zip](https://github.com/user-attachments/files/15754207/hk_dev.bin.zip)

x23piracy, 2024-06-09:
Hi, thanks so far but it get: 

Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)