rednblkx/HomeKey-ESP32 #64: Installation of 2 HomeKit Locks

Spikey85, 2024-06-14:
Good morning, I would like to install 2 Homekey readers, one at the front door and one at the apartment door to open each individual door. At the apartment door with Nuki and Home Assistant, the automations work perfectly. If I now install the second one, I have the problem that the "reader" (ESP32 and PN532) in HA are the same MQTT device, the same TAGS (device_id) and there is only one Homekey card in Wallet. I can't manage to separate the automations in HA and only control one device, even with scanned_tags or something like that. I changed the name in config.h to #define MQTT_CLIENTID & #define Name. But both always respond. I also changed the serial number and device ID etc. in the depths of your project. In HA in MQTT they are now also displayed as 2 devices. But the device IDs for the TAGs are the same.

Is it because both devices use the same Homkey card in Wallet? Or what can I do to ensure that both readers are recognized as completely different devices and work as such?

I hope someone can help me, I'm desperate :-)

Best regards,
Stefan

rednblkx, 2024-06-14:
Hello!

In order to deploy multiple devices flashed with this, you only need to do two things

1. Change `MQTT_CLIENTID` to something unique or the device won't be able to join MQTT
2. Modify the topics in the "MQTT Topics" section to something else as that's how you end up with this mangling, it's enough to change the first part, e.g. "topic/custom_state" -> "another_topic/custom_state"

The name is only used for display purposes and the serial number it's derived from the mac address of the device so it is already unique.

Apple designed the HomeKey to be one per Home in order to simplify things as it doesn't make any sense to have multiple "digital keys" for each lock within the same house and makes it easier to identify devices.

Try the steps and let me know how it went, if it doesn't work, you might need to cleanup your MQTT setup.

Spikey85, 2024-06-14:
Wow, you're my hero. It works perfectly. Unfortunately, I'm not that good at MQTT yet.

Thank you for the quick help and for making this possible.

rednblkx, 2024-06-14:
Great, happy to help!

closed 2024-06-14 by rednblkx