rednblkx/HomeKey-ESP32 #65: Wifi -- why so hard?

segdy, 2024-06-15:
Phew, surprising I am getting stuck here at connecting to Wifi.... Usually there's an AP for initial configuration or at least configuration at compile time.

I am just re-using an ESP32-CAM module to play around so I don't have a switch.

Flashing the compiled firmware works well but how the heck do I get access to the serial console? I tried connecting to the COM port via putty, tried 9600, 115200, no flow control, 8n1 but I either get nothing or just a few characters of garbage. 

Is there any straight forward way to get connected to the Wifi so I can just play around with the HAP part a bit?

konradhuebner, 2024-06-15:
@segdy it does work with 115200 on the serial port. After connecting, reset the device so that it runs through a clear boot. I used CoolTerm on MAC. If you then press "A" it does start a local hotspot

segdy, 2024-06-15:
Thank you. Still no luck :-(

[image]

[image]

Pressing reset button does nothing. Unfortunately I can't open the serial console from boot on because once I unplug USB, the serial port is gone. So after plugging in I have to wait a fraction of time to open the serial port. When I success I get the screenshot above.

Again, it flashed successfully so it can't be completely off.

rednblkx, 2024-06-16:
that looks like either wrong baud rate or the serial bridge is a bit funky.
The correct baud rate is 115200 and is defined [here](https://github.com/rednblkx/HomeKey-ESP32/blob/3b05ef95dde92fe378c62623dbb2029560f8a9e4/src/main.cpp#L1149)

Try the new version, now the access point autostarts at boot.

segdy, 2024-06-16:
Absolutely amazing!! So much just happened :-) :-) 

I tried to build the newest version but I am getting tons of errors now. I updated platform IO and I also cleaned everything and push'ed the latest version:

```
[...]
Compiling .pio\build\release\bootloader_support\src\esp_image_format.c.o
src/main.cpp:59:37: error: 'MQTT_DISCOVERY' was not declared in this scope
     bool hassMqttDiscoveryEnabled = MQTT_DISCOVERY;
                                     ^~~~~~~~~~~~~~
src/main.cpp:59:37: note: suggested alternative: 'DISCOVERY'
     bool hassMqttDiscoveryEnabled = MQTT_DISCOVERY;
                                     ^~~~~~~~~~~~~~
                                     DISCOVERY
Compiling .pio\build\release\bootloader_support\src\flash_encrypt.c.o
src/main.cpp:66:30: error: 'DEVICE_NAME' was not declared in this scope
     std::string deviceName = DEVICE_NAME;
                              ^~~~~~~~~~~
src/main.cpp:68:29: error: 'SETUP_CODE' was not declared in this scope
Compiling .pio\build\release\bootloader_support\src\secure_boot.c.o
     std::string setupCode = SETUP_CODE;
                             ^~~~~~~~~~
src/main.cpp:68:29: note: suggested alternative: 'pdTASK_CODE'
     std::string setupCode = SETUP_CODE;
                             ^~~~~~~~~~
                             pdTASK_CODE
src/main.cpp:69:29: error: 'HOMEKEY_ALWAYS_UNLOCK' was not declared in this scope
     bool lockAlwaysUnlock = HOMEKEY_ALWAYS_UNLOCK;
                             ^~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\release\bootloader_support\src\flash_partitions.c.o
src/main.cpp:69:29: note: suggested alternative: 'MQTT_HOMEKEY_ALWAYS_UNLOCK'
Compiling .pio\build\release\bootloader_support\src\flash_qio_mode.c.o
     bool lockAlwaysUnlock = HOMEKEY_ALWAYS_UNLOCK;
                             ^~~~~~~~~~~~~~~~~~~~Compiling .pio\build\release\bootloader_support\src\bootloader_flash_config_esp32.c.o
~
                             MQTT_HOMEKEY_ALWAYS_UNLOCK
src/main.cpp:70:27: error: 'HOMEKEY_ALWAYS_LOCK' was not declared in this scope
     bool lockAlwaysLock = HOMEKEY_ALWAYS_LOCK;
                           ^~~~~~~~~~Compiling .pio\build\release\bootloader_support\src\bootloader_efuse.c.o
~~~~~~~~~
src/main.cpp:70:27: note: suggested alternative: 'MQTT_HOMEKEY_ALWAYS_LOCK'
     bool lockAlwaysLock = HOMEKEY_ALWAYS_LOCK;
                           ^~~~~~~~~~~~~~~~~~~
                           MQTT_HOMEKEY_ALWAYS_LOCK
Compiling .pio\build\release\bootloader_support\src\idf\bootloader_sha.c.o
src/main.cpp:71:26: error: 'HS_PIN' was not declared in this scope
     uint8_t controlPin = HS_PIN;
                          ^~~~~~
src/main.cpp:71:26: note: suggested alternative: 'WS_PING'
Compiling .pio\build\release\cbor\tinycbor\src\cborencoder_close_container_checked.c.o
     uint8_t controlPin = HS_PIN;
                          ^~~~~~
                          WS_PING
src/main.cpp:72:27: error: 'HS_STATUS_LED' was not declared in this scope
     uint8_t hsStatusPin = HS_STATUS_LED;
                           ^~~~~~~~~~~~~
src/main.cpp:72:27: note: suggested alternative: 'HS_STATUS'
     uint8_t hsStatusPin = HS_STATUS_LED;
Archiving .pio\build\release\esp-idf\bootloader_support\libbootloader_support.a
                           ^~~~~~~~~~~~~
                           HS_STATUS
src/main.cpp:73:29: error: 'NFC_SUCCESS_PIN' was not declared in this scope
     uint8_t nfcSuccessPin = NFC_SUCCESS_PIN;
                             ^~~~~~~~~~~~~~~
Compiling .pio\build\release\cbor\tinycbor\src\cborencoder.c.o
src/main.cpp:74:31: error: 'NFC_SUCCESS_TIME' was not declared in this scope
Compiling .pio\build\release\cbor\tinycbor\src\cborerrorstrings.c.o
     uint16_t nfcSuccessTime = NFC_SUCCESS_TIME;
                               ^~~~~~~~~~~~~~~~
src/main.cpp:75:25: error: 'NFC_SUCCESS_HL' was not declared in this scope
[... 46 lines cut ...]
     bool gpioActionLockState = GPIO_ACTION_LOCK_STATE;
                                ^~~~~~~~~~~~~~~~~~~~~~
                                GPIO_CALI_START
src/main.cpp:82:34: error: 'GPIO_ACTION_UNLOCK_STATE' was not declared in this scope
     bool gpioActionUnlockState = GPIO_ACTION_UNLOCK_STATE;
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:82:34: note: suggested alternative: 'GPIO_PIN_INT_TYPE'
     bool gpioActionUnlockState = GPIO_ACTION_UNLOCK_STATE;
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
                                  GPIO_PIN_INT_TYPE
src/main.cpp: In function 'void setup()':
src/main.cpp:1250:28: error: 'DEVICE_NAME' was not declared in this scope
   new Characteristic::Name(DEVICE_NAME);
                            ^~~~~~~~~~~
Compiling .pio\build\release\coap\libcoap\src\address.c.o
Archiving .pio\build\release\esp-idf\cmock\libcmock.a
Compiling .pio\build\release\coap\libcoap\src\async.c.o
Compiling .pio\build\release\coap\libcoap\src\block.c.o
Compiling .pio\build\release\coap\libcoap\src\coap_asn1.c.o
Compiling .pio\build\release\coap\libcoap\src\coap_cache.c.o
*** [.pio\build\release\src\main.cpp.o] Error 1
[...]
```

Meanwhile I downloaded your repcompiled binary and flashed if via esptool.py. I could eventually connect my Wifi, add it to Home app and now the key is already showing up in my wallet :-) :-)  So awesome. I'm stoked and looking forward for the NFC reader to arrive, but this looks already so promising. Thanks for the great work.

rednblkx, 2024-06-17:
Great, happy to hear, If you have any questions, you can write me an email, join the discord server or open a discussion.

closed 2024-06-17 by rednblkx