rednblkx/HomeKey-ESP32 #72: Didn't find PN53x board

IvanDeMarco, 2024-06-22:
Hi,
I have the error "Didn't find PN53x board".
PN532 is correctly connected with LED on. Wiring is correct according guide.
I have installed the latest version of the software.
I'll post some photos to show you the connection:
[image]
[image]
[image]

I managed to integrate it into home assistant via mqtt and homekit, I generated the key but clearly it doesn't work...
Please help me! What I'm doing wrong?

rednblkx, 2024-06-22:
Hi there!

Yeah, there seems to be an issue with some of the nfc boards that hasn't yet been identified, though not 100% confirmed it's just a hardware issue and unfortunately don't yet have a board with this issue.

In the meantime: 
 1. If you were powering from a laptop(especially a mac), try powering from the wall and just see if the iphone reacts
 2. try swapping the wires since signal integrity is a thing.

I'll let you know when a fix is available

IvanDeMarco, 2024-06-22:
Thanks for the reply.
In what sense try to swap the wires?
If you can tell me which card to buy that definitely works through a link I will buy the product you indicate.

Thanks for the support!

rednblkx, 2024-06-22:
It’s not yet confirmed what is the root cause of this so i can’t recommend a board to buy. 
On the latest version i did a slight refactor of the driver and someone who updated to it reported the same issue while it worked on the previous so there might be a timeout issue or something with some boards.

If you can, compile the previous [0.2](https://github.com/rednblkx/HomeKey-ESP32/archive/refs/tags/v0.2.zip) version and let me know if it works or I’ll drop you the firmware file later.

> In what sense try to swap the wires?

Change the wires with other ones, it’s stupid but there is a chance.

rednblkx, 2024-06-26:
Hi again,

So it seems this might be a hardware issue, someone on the Discord server had this issue and it turned out to be a cable length issue, they were using 1m, max would be like 15cm, doesn’t look like that would be the case here but still could be a connection issue, make sure all DuPont connectors are plugged in all the way, if some are loose (the one connected to MOSI looks a bit loose) then use a different wire with a more snug connector, or solder the wires directly just to be sure.

Let me know if anything.

IvanDeMarco, 2024-07-09:
hi, I checked the continuity of the electrical conduction with the tester and everything is correct. I would like to understand what possible solutions there may be, if it works for you it should work for me too if I buy the same product... thanks a lot!!

IvanDeMarco, 2024-07-11:
I tried changing all the cables (mine are 10cm) and nothing changed. Can you share the Discord channel so we can follow the topic? However, I would never hold you responsible for the purchase if the new card doesn't work so feel free to share the product.

Thank you!

rednblkx, 2024-07-11:
https://discordapp.com/invite/VWpZ5YyUcm

closed 2024-07-27 by rednblkx

bananajoe86, 2024-08-12:
in my case I got a Dip-Switch. I choose SPI (lower Swich2 to on) an it work. 
[image]