rednblkx/HomeKey-ESP32 #74: State not changing in Home App

mkomarac, 2024-06-27:
Hi.
First of all, i must say it is great project. and it works almost perfectly.
the issue is when using lock inside Home app when changing state from Locked towards Unlocked, it says "unlocking..." and the state never changes,
 
but if I change state in Home Assistant, from lock to unlock, the change is momentary like in Home Assistant and in Home app to correct state; 
And again, if want in Home app change state from unlocked towards locked, it stays "locking..." forever, and never change state

What is the solution?

thanks in advance

rednblkx, 2024-06-28:
Hi there,

First of all, the project relies on MQTT to actually be useful and fully working, though recently i implemented some rudimentary GPIO control.

The lock on the Home app is more of a dummy one as it is not really connected to anything, otherwise it just sends target state change request from the Home App to the "State Topic" and stays in the "Locking" or "Unlocking" state until the current state is updated through either the "Current State Cmd Topic" or the "State Cmd Topic".

Interacting with the lock from HASS does exactly the above as that's the purpose of the MQTT Topics and so will not get stuck in a "locking..." state.

All this is explained on the wiki in the [MQTT](https://github.com/rednblkx/HomeKey-ESP32/wiki/MQTT) section, though i admit, could be better and i'm working on improving the documentation.

Let me know if you encounter any other issues or have any questions. You can also join the Discord server if you want where we discuss about homekey, smart home stuff and anything in between.

mkomarac, 2024-06-28:
If I understand correctly:

When I scan HomeKey generated in home app on the PN532 attached to ESP32 running this code, it will first trigger Home Assistant lock entity, and then reflect this state towards Home App?

I have tried to scan HomeKey on another NFC reader attached to home assistant, and each time I scan HomeKey form Apple Wallet, it looks like I am scanning every time new card (or NFC tag)

What's the right way to use HomeKey from Apple wallet?

If I understand correctly: HomeKey generated in iOS Wallet, is "paired" to code running inside ESP32, and PN532 is just interface to check and match HomeKey and "lock"

If that's true, when I scan HomeKey from Apple Wallet on PN532 attached to this ESP32 running this code, it should show up in tags as unique key?
is that correct workflow?

rednblkx, 2024-06-30:
> When I scan HomeKey generated in home app on the PN532 attached to ESP32 running this code, it will first trigger Home Assistant lock entity, and then reflect this state towards Home App?

In a sense yes, HomeKit has a current-target state for locks and when interacting with it on the Home App it only sets the "target state" which this project publishes over MQTT on the State Topic that you set on the WebUI and in order for the cycle to be complete, the "current state" has to then be set to the same value(0=unlocked or 1=locked) using one of MQTT control topics of this project.

> I have tried to scan HomeKey on another NFC reader attached to home assistant, and each time I scan HomeKey form Apple Wallet, it looks like I am scanning every time new card (or NFC tag)

That's because Home Key is not meant to be used as a regular NFC Tag, it's only emulating one in order to communicate with readers and it's generating a new UID everytime for security purposes.

> What's the right way to use HomeKey from Apple wallet?

The "right way" is to use it with the PN532 attached to the ESP32 flashed with this project which performs the cryptographic exchanges Home Key expects.

> If I understand correctly: HomeKey generated in iOS Wallet, is "paired" to code running inside ESP32, and PN532 is just interface to check and match HomeKey and "lock"
>
> If that's true, when I scan [...]

Mostly yes, once the HomeKey is successfully authenticated, two IDs are resulted, the `issuerId` which is unique for the icloud acc and an `endpointId` which is unique per device, for this reason you will actually see two "tags", one that identifies you and one that identifies the device, the longer ID is the issuerId.

closed 2024-07-16 by rednblkx