rednblkx/HomeKey-ESP32 #81: Topic homekey_mqtt/status?

segdy, 2024-07-06:
I just got my hardware and it works super well. YAY!

However, HomeAssistant shows the lock as unavailable:

[image]

I did check "Enable HASS MQTT Discovery". I think the issue is the availablity_topic in the auto disccovery which is always set to topic "homekey_mqtt/status":

[image]

But this topic doesn't exist and I didn't define it anywhere. Is this a bug? It should be in homekey/homekey/status

segdy, 2024-07-08:
Removing availability_topic doesn't actually work because the topic will be overwritten.
As another temporary fix, I just created retained topic `homekey_mqtt/status` with value `online`. This works, but fix in code would be highly appreciated!

rednblkx, 2024-07-08:
Hi there!

`"homekey_mqtt/status"` is an internal topic defined on connection, used for [LWT](https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/) and also as the `availability_topic` for HASS as that is required.

If this topic is not present, you might have connectivity issues which prevents the esp from telling the broker it's "alive" or more specifically setting `status` to `online`.

In normal conditions this works without issues, suggest to check the mqtt settings on the esp, try changing the client id, perhaps there might be a conflict for some reason.

segdy, 2024-07-08:
I'm confused. The topic should be configured somewhere, just as the other ones are configured:

[image]

There is nowhere anything to configure the availability topic (or availability function in the first place).

Instead of "homekey_mqtt/status" it should be "homekey/homekey/availability" (or similar). And HomeKey-ESP32 (HomeSpan) is supposed to write this ... but doesn't.

rednblkx, 2024-07-08:
I haven't included it as an option since it doesn't add any benefit to being configurable as it really was only meant to be used by HASS and it is also made to be "unique" as it uses the mqtt client id for the first part.

segdy, 2024-07-08:
Yes, I see what's going on now. "$clientID/status" topic is being generated automatically and used as availability topic. Most often, users have fine grained access control (incl myself), so access to topic "homekey_mqtt/" was not allowed b y mosquitto ACL. I can make it work by changing Client ID to "homekey". So, there are many workarounds (three now described in this thread).

While not something critical, I'd still think it would be a good idea to make this configurable. Because:

> since it doesn't add any benefit to being configurable as it really was only meant to be used by HASS

with the same argument you wouldn't need anything configurable. The benefit of letting users change all the MQTT topics is exactly identical to letting them change the availability topic. (see my story above as an example).

The behavior that a device automatically generates an availability topic under "$clientID" I have not experienced with any other device so far. I don't think it's common or best practice. Just as an example, OpenGarage uses an automatically generated client ID derived from MAC address (OG_B02E82 in my example). Yet, the availability topic shows up as "/garage/OUT/STATUS" and not "OG_B02E82/STATUS".

rednblkx, 2024-07-10:
Just to be clear, the project is not yet in final shape, main page and versioning I think says as much.
I’m not saying things like this will stay as they are, just providing some reasoning to why it is how it is now, i did not thought it was worth the extra lines at the time so I did not include it. 
Note that it is mentioned in the [wiki](https://github.com/rednblkx/HomeKey-ESP32/wiki/HASS) that it uses `<MQTT_CLIENTID>/status` as the topic for LWT though I agree I should’ve mentioned it is also used in `availability_topic`.

On a side note, this is exactly why I enabled the Issues and Discussions tab, so I know what things need to be worked on as I try to make the project as good as it can be.

Thank you for letting me know, it will be handled.

rednblkx, 2024-10-08:
Added in 0.4

closed 2024-10-08 by rednblkx