rednblkx/HomeKey-ESP32 #84: MQTT auto discovery with multiple instances

skrebber, 2024-07-16:
Hi,
thanks for your ongoing work with this fantastic project! Really appreciate!

I installed your software to two esp's with each connected to a pn532, configuring two separate locks with different names in the web-ui. 
Home assistant recognizes the first in all matters via mqtt auto discovery. I noticed that the second one only creates a device but no entities. I looked into the mqtt broker and saw that you don't give individual names to the auto discovery topic:
HA-MQTT manual states the following syntax: 
`<discovery_prefix>/<component>/[<node_id>/]<object_id>/config`

Your code creates twice this topic but with individual config-json.
`homeassistant/lock/homekey_mqtt/lock/config`
I think the second (last) written lock should have the individual name of the lock-device. Manually sending each config-json created by your code to two separate topics makes HA autodiscover work:

```
homeassistant/lock/homekey_mqtt/frontdoor/config
homeassistant/lock/homekey_mqtt/backdoor/config
```

I hope it helps your project.
Best regards.

rednblkx, 2024-07-16:
Hi there!

Thanks for letting me know, appreciate it!

It’s actually intended behavior, only the wiki lacks some info atm, bear with me.

In order to deploy multiple devices within the same network you only need two things, but one in particular, change “Client ID”(important) in “MQTT Config” and also “Device Name” in Misc.

The `homekey_mqtt` part of `homeassistant/lock/homekey_mqtt/lock/config` is given by the “Client ID” you set in “MQTT Config”, if you set it to something it will then be a different topic.

Let me know if that helped.

skrebber, 2024-07-17:
Hi rednblkx,
ok, now I understand your approach. And yes, changing the client-ID helped. But one question further: I assume that in the mqtt core topics I should also configure and use different "topic"s to divide the mqtt communication of the two locks. 
Thanks and have a nice day!

rednblkx, 2024-07-17:
Yes, that’s right

closed 2024-07-17 by skrebber