rednblkx/HomeKey-ESP32 #87: How to share apple home key?

yannik2480, 2024-07-21:
Hey, while setting everything up, I ran into difficulty sharing Apple HomeKey with my family members. I tried adding them in the settings but couldn’t find any “add” button. I'm not using an Apple gateway like Apple TV or HomePod. Does anyone know a solution? Is it because I'm not using a gateway, and if so, how can I use it without one?

ihrapsa, 2024-07-21:
You need to tap the “+” button that you usually tap when adding accessories in Home app. You’ll see “Add people” there.

yannik2480, 2024-07-21:
[image][image]

ihrapsa, 2024-07-21:
That’s weird. Seeing now that lots of people online have this issue. Some say that you need a hub, however there’s no clear apple documentation. If you read Apple docs[here](https://support.apple.com/en-eg/102386) at the beginning it says that `You need a hub to share control of home`. However, at the end it says that  `If you don't have a home hub set up, they can only control your accessories while they're at your home, connected to your home Wi-Fi network`
What iOS version are you on?

ihrapsa, 2024-07-21:
Ok, I had to read to the end:
You need to be on Home Network to invite people.
[image]

rednblkx, 2024-07-21:
That makes sense, looking at the screenshot they don’t seem to be connected to a WiFi.
I was starting to think Apple may have started forcing the new HomeKit Architecture on everyone so we’ll have to buy HomePods 😅

yannik2480, 2024-07-21:
So, if I understood correctly, I need to be connected to my WiFi network with my iPhone. I have tried that, but the same issue occurs.

rednblkx, 2024-07-21:
Then unfortunately you need a HomePod or Apple TV, it looks like you are on the new HomeKit architecture which requires a home hub.

yannik2480, 2024-07-22:
I used the following to downgrade which worked!

I also had the issue where I have no Home app option in settings.  It turns out you have to do the following:
click one this link in safari http://appldnld.apple.com/iOSProfiles/KeepLegacyHome.mobileconfig and follow those instructions 
go to settings > general > VPN (because why wouldn’t it be there?? So stupid) and you should see something about the profile thing you installed (I forget what it says). It will say something along the lines of “installs a button to reset Home configuration” 
Follow those steps, THEN you’ll magically have a Home option in settings again. 
Go to the heretofore MIA Home option in your phones settings and tap the reset home configuration, or whatever it says.  
That’s as far as I’ve gotten but I wanted to post this before I lost the motivation to do so.

closed 2024-07-23 by rednblkx

werty1st, 2024-09-16:
Is it not possible to add an other Card/User to the esp32?
(We already open the door with HA but I would like to also use the Apple Wallet.)

yannik2480, 2024-09-17:
> Is it not possible to add an other Card/User to the esp32? (We already open the door with HA but I would like to also use the Apple Wallet.)

It is possible to share the card, but only in the apple home app. Since the new architecture doesn’t support sharing a home anymore you would need to downgrade to the old architecture (see above). Or you need to buy an apple homepod.

I’m not sure if it’s supported but somewhere I read that physical nfc cards are also working, not sure if that’s true or how to pair them…

rednblkx, 2024-09-17:
> Is it not possible to add an other Card/User to the esp32? (We already open the door with HA but I would like to also use the Apple Wallet.)

As the Key in the Wallet is tied to the Home App, there is no possibility of sharing from the ESP. If you don’t have the option for invite, just follow the guide above that @yannik2480 kindly shared.

The project integrates with HA through MQTT and the HomeKey and other scanned Tags will appear in the Tags Section of HASS.

ImNoahDev, 2025-01-13:
That guide does not seem to work any longer. The link is invalid

yannik2480, 2025-01-13:
> That guide does not seem to work any longer. The link is invalid

Are you sure? The link is still working fine for me (on iPhone)

ImNoahDev, 2025-01-14:
It shows as `not found on accelerator ` when loaded. 
I pulled the link from the wayback machine but it says the profile is not supported on iOS 18

ImNoahDev, 2025-07-23:
It was still working for me not long ago. You need to use HTTP and not HTTPS.
Try: http://appldnld.apple.com/iOSProfiles/KeepLegacyHome.mobileconfig