rednblkx/HomeKey-ESP32 #89: Reconnect to MQTT Broker

hjaltij, 2024-07-23:
Hi,

I've noticed that sometimes the device loses connection to the MQTT broker and reading through the code I don't see any attempts to try to reconnect. This makes it so you have to restart the ESP to regain connection. Is there any work being done to detect disconnects and attempt to reconnect?

Thanks for a great project!

hjaltij, 2024-07-23:
Digging deeper, perhaps the MQTT library is supposed to handle this automatically?

What I'm seeing sometimes after leaving the device for a some time and coming back I get the following errors.

`E (36555749) esp-tls: [sock=50] select() timeout
E (36555750) TRANSPORT_BASE: Failed to open a new connection: 32774
E (36555750) MQTT_CLIENT: Error transport connect`

rednblkx, 2024-07-23:
Hi,

Yes, the component that handles MQTT will attempt to reconnect when it detects disruptions. If you are experiencing frequent disconnections you might have some connectivity issues.

Is the ESP far from the AP? Do you perhaps have it behind something that could cause interference, something metal?

hjaltij, 2024-07-23:
It's about 40cm from my AP since I'm still doing some work on this on my desk. It's also not in any enclosure yet.

I'll do some more debugging and see if I can figure out why it's dropping the connection. FWIW I still just have it running over USB from my desktop and watching the serial output. Not sure if there's something that causes it to drop when my machine sleeps.

rednblkx, 2024-07-26:
on sleep, it switches to the standby rail and it possibly confuses the WiFi chip.

hjaltij, 2024-07-26:
Makes sense. I guess we can close this one until I've tested it over longer perioid with a stable power source (not my desktop).

Thanks!

closed 2024-07-26 by hjaltij