rednblkx/HomeKey-ESP32 #90: QUESTION: Possible usage of ESP32-C3 SuperMini (e.g. Seeed Studio XIAO ESP32C3)

niiccooo1, 2024-07-24:
Hello,

first of all thank you for this nice project. I would like to implement this under housing of my Ring Doorbell camera so my space is limited. The above mentioned ESP32-C3 wolud fit but I'm not sure if the communcation between ESP32 and PN532 work.
The Seeed Studio XIAO ESP32C3 pinout shows only MISO, MOSI and SCK. It is missing SS, or?
[image]

Thank in advance!

Kind regards.

rednblkx, 2024-07-25:
Hello 👋!

Thanks for your interest.

Don't know why they marked 10,9,8 as SPI Pins but it doesn't matter really as the project doesn't use native(IO_MUX) pins and so any pin can be used provided you assign them accordingly in the code.

As noted in the wiki, the pins for C3 are `SS: 7, SCK: 4, MISO: 5, MOSI: 6`, which as i can see are available on the board, try it and let me know how it goes.

Haven't tried this board so unfortunately don't have any insight on it.

Best regards

niiccooo1, 2024-07-26:
Thanks for your reply. I ordered some ESP32C3. I'll try and give a feedback.

DNGRSDave84, 2024-07-26:
hi just tested, works on the ports as [rednblkx](https://github.com/rednblkx) said  , SS: 7, SCK: 4, MISO: 5, MOSI: 6

closed 2024-07-26 by niiccooo1

adukale, 2024-08-18:
If someone stumbles upon this thread and still get "NFC_SETUP: Didn't find PN53x board" after using above mentioned pins, then try using below pins.
Using below pins I am able to establish the communication with Seeed Studio XIAO ESP32C3. Apparently there are HSPI and VSPI pins. The below pins are HSPI pins which are default SPI pins. Cheers.
SS    = 5;
MOSI  = 4;
MISO  = 3;
SCK   = 2;

TiddlyWiddly, 2025-06-07:
@adukale I was running into the "NFC_SETUP" error that you pointed out. I moved wiring to 5,4,3,2 as described and also changed pins in webUI but still received the error. I kept wiring on 5,4,3,2 but used default pins 7,6,5,4 in web UI and I don't get an error, but I don't think this is right.  I see HK in my Home app, but trying to add says "Unable to Add Accessory". No error logs on console. Did the pins match in the UI?