rednblkx/HomeKey-ESP32 #95: reboot loop

Maxime0506, 2024-08-03:
reboot loop and unresponsive and this is what i get in the serial monitor:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5596
ho 0 tail 12 room 4
load:0x40078000,len:15852
load:0x40080400,len:3624
entry 0x4008064c
[0;32mI (591) cpu_start: Multicore app[0m
[0;32mI (591) cpu_start: Pro cpu up.[0m
[0;32mI (592) cpu_start: Starting app cpu, entry point is 0x40082138[0m
[0;32mI (580) cpu_start: App cpu up.[0m
[0;32mI (610) cpu_start: Pro cpu start user code[0m
[0;32mI (610) cpu_start: cpu freq: 240000000[0m
[0;32mI (610) cpu_start: Application information:[0m
[0;32mI (614) cpu_start: Project name:     HomeKey-ESP32[0m
[0;32mI (620) cpu_start: App version:      1[0m
[0;32mI (624) cpu_start: Compile time:     Jun 30 2024 23:56:39[0m
[0;32mI (630) cpu_start: ELF file SHA256:  14e48647b84b2abf...[0m
[0;32mI (636) cpu_start: ESP-IDF:          4.4.6[0m
[0;32mI (641) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (646) cpu_start: Max chip rev:     v3.99 [0m
[0;32mI (650) cpu_start: Chip rev:         v3.1[0m
[0;32mI (655) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (663) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (668) heap_init: At 3FFBB0D8 len 00024F28 (147 KiB): DRAM[0m
[0;32mI (675) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (681) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (688) heap_init: At 40098B34 len 000074CC (29 KiB): IRAM[0m
[0;32mI (695) spi_flash: detected chip: generic[0m
[0;32mI (698) spi_flash: flash io: dio[0m
[0;32mI (707) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (7
***ERROR*** A stack overflow in task loopTask has been detected.

Backtrace: 0x40082aee:0x3ffc0640 0x40089cb5:0x3ffc0660 0x4008d546:0x3ffc0680 0x4008bb54:0x3ffc06f0 0x40089db0:0x3ffc0710 0x40089d62:0x00000003 |<-CORRUPTED

ELF file SHA256: 14e48647b84b2abf

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5596
ho 0 tail 12 room 4
load:0x40078000,len:15852
load:0x40080400,len:3624
entry 0x4008064c

rednblkx, 2024-08-04:
Judging by the fact that it looks to have crashed upon starting the scheduler and you also have a stack overflow on the loop task I say you have a bad chip which is something I’ve seen happen, you have to keep in mind that most dev boards you buy unless they are from Espressif or a known good manufacturer, they are just random Chinese boards with an ESP32 chip that only god knows what level of QA actually went through.

I have a chip where the WiFi has just started to act up, it either occasionally crashes when trying to connect or i can’t connect to the AP and it was fine before.

Maxime0506, 2024-08-04:
i have one from AZ Delivery a german brand

rednblkx, 2024-08-05:
ok, i won't debate over that, the point is that in normal conditions this issue would not occur and based on the log you sent it looks like a hardware issue.

How are you powering the board?

closed 2024-08-15 by rednblkx