rednblkx/HomeKey-ESP32 #98: Not responding as soon as added in Home app

davidgrdm, 2024-08-05:
Hi everyone, 
As soon as I add my ESP32 to the Home app on iOS, it became unresponsive  "Not responding"
[image]
.
I tried with two different supported ESP32 devices and the issue is the same.
As anyone encountered the same issue ?

I tried on iOS 17.5 and iOS 18.0 beta devices.

yassik31, 2024-08-05:
Same problem for me

davidgrdm, 2024-08-07:
Okay, in case someone have the same issue : HomePod is causing the issue. Unplugged it, so the Apple TV is the hub, reflashed and everything works perfectly :).

closed 2024-08-07 by davidgrdm

Antti, 2024-09-06:
I don't have a HomePod, but still have the same issue. I have to restart Apple TV for it to start working.
Clearly, it's not a HomePod or Apple TV is causing the issue, it's a problem somewhere in this software (most likely in Homespan).