rednblkx/HomeKey-ESP32 #102: Hardware Lock Time (ms)

DarkFiguringitout, 2024-08-08:
First of all, great project and well written up guide. Got everything hooked up and working quite quickly. 

However, I am currently using HomeKey Events trigger on successful authentication. My system requires the button to be temporarily be pulled down. All good on authentication locally, however remotely over home assistant or the home app, this doesn't help me.

Could you help me with what I have to modify to add a temporary trigger to the hardware lock too? I really just need it to be pressed shortly, and go into a locked state after that.

Thank you!

rednblkx, 2024-08-08:
Hi!

I'm glad the guide was useful.

Currently, there is no momentary state implemented, but i'm planning on including that with the next version.

Thank you for your interest!

fuelle, 2024-08-11:
I solved that, by using a timer relay (https://amzn.eu/d/b8mszOj) which is closing for 5 seconds.
An automation in HomeKit "locks" the door, as soon as it is unlocked, so the setting "Always Unlock on HK Auth" will always work.

rednblkx, 2024-10-08:
Added in 0.4

closed 2024-10-08 by rednblkx