rednblkx/HomeKey-ESP32 #106: Unable to connect to wifi

fschaal, 2024-08-13:
Hi, I've seen a few issues around this but for me, nothing is working. I get the captive portal but whatever wifi I use would not connect staying on waiting x seconds to check for response.

I've already tried some different boards like those mentioned in other issues. I've even created a brand-new test wifi with only 2.4gh but that did also not connect.

I've used this command to flash: `esptool.py --port /dev/cu.usbserial-0001 write_flash 0x0 ~/Desktop/esp32-firmware_merged.bin`

 Output:
```
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:d1:f9:cf:69:c8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x003fffff...
Compressed 4194304 bytes to 1068210...
Wrote 4194304 bytes (1068210 compressed) at 0x00000000 in 98.4 seconds (effective 341.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

What am I missing?

rednblkx, 2024-08-15:
Hi!

If you are sure you provided the right credentials then it is possibly you have a faulty board as that seems to be a common issue.

rednblkx, 2024-08-15:
I understand you also tried different boards but at this point, unfortunately I do not have other suggestions, it’s a rare issue and makes it harder to identify what’s triggering it since there is no specific error or anything.

closed 2024-08-26 by rednblkx