rednblkx/HomeKey-ESP32 #108: FlowFailed

alpdmrel, 2024-08-14:
I keep getting this error. How can I fix it ?

[18:32:17]I (15389) MAIN: nfc_thread_entry > *** PASSIVE TARGET DETECTED ***
[18:32:17]I (15605) HKAuthCtx: HKAuthenticationContext > Initialization Time: 188 ms
[18:32:17]E (15661) HKAuthCtx: authenticate > Response not valid, something went wrong!
[18:32:17]W (15662) MAIN: nfc_thread_entry > We got status FlowFailed, mqtt untouched!

rednblkx, 2024-08-15:
Please try to add it again in the Home app.

alpdmrel, 2024-08-16:
I tried it multiple times with differnt boards but the same problem exists.
I also switched to 5cm cables and differnet power supply but nothing worked.

rednblkx, 2024-08-16:
Please use the `@LD` command to get more detailed logs, don't wait for an end, watch out for the error or wait about 5-10 sec as this will also print logs from the pn532 driver and that includes the loop at the end that checks for presence

rednblkx, 2024-08-16:
Also, are you on the iOS 18 Beta?

alpdmrel, 2024-08-16:
I am on iOS 18 beta 6 I also tried on beta 4.
Here is the logs with @LD https://pastebin.com/RKxuTgK2

rednblkx, 2024-08-16:
That might be an issue, heard mixed results with iOS 18.

Logs you posted are incomplete, what are you using as serial monitor? Try https://serial.huhn.me

Also, please enclose any text starting with `@` in backticks (`` `@LD` ``), that is so you don’t ping random people.

rednblkx, 2024-08-16:
Ah nvm, I see now you edited

rednblkx, 2024-08-16:
Ok, i see what the main issue is, the iPhone should provision the “lock” with its private key during pairing but it seems that it did not.

Could you also share logs from during pairing?

alpdmrel, 2024-08-16:
Here is the logs during pairing

```
D (72413) MAIN: pairCallback > Found allocated controller - Hash: DF725617D9CB5A0D[0m

D (72414) MAIN: pairCallback > Adding new issuer - ID: DF725617D9CB5A0D[0m

D (72424) MAIN: save_to_nvs > NVS SET STATUS: ESP_OK[0m

D (72424) MAIN: save_to_nvs > NVS COMMIT STATUS: ESP_OK[0m

D (99281) NFCAccess: update > PROVISIONED READER KEY: [0m

D (99282) NFCAccess: update > READER GROUP IDENTIFIER: [0m

D (99282) NFCAccess: update > READER UNIQUE IDENTIFIER: [0m

D (99287) NFCAccess: update > Decoded data: 010102045201010202404A4D390F76BB0A2BD907C82DBDDDD849ABFEF173567FD3C4FEE95B70B1A0D9B7503794ED81A7AA4861C5F1CEA72A26A2452544DA73AF7D57E095F3A54DFF82A50308DF725617D9CB5A0D040101[0m

D (99306) NFCAccess: update > Decoded data length: 87[0m

[0;32mI (99311) HK_HomeKit: processResult > TLV OPERATION: 2[0m

[0;32mI (99317) HK_HomeKit: processResult > TLV DCR: 1[0m

D (99322) HK_HomeKit: processResult > PROVISION DEVICE CREDENTIAL REQUEST[0m

D (99329) HK_HomeKit: provision_device_cred > DCReq Buffer length: 82, data: 01010202404A4D390F76BB0A2BD907C82DBDDDD849ABFEF173567FD3C4FEE95B70B1A0D9B7503794ED81A7AA4861C5F1CEA72A26A2452544DA73AF7D57E095F3A54DFF82A50308DF725617D9CB5A0D040101[0m

D (99352) HK_HomeKit: provision_device_cred > Found issuer - ID: DF725617D9CB5A0D[0m

D (99359) HK_HomeKit: provision_device_cred > Adding new endpoint - ID: 79D642E6727E , PublicKey: 044A4D390F76BB0A2BD907C82DBDDDD849ABFEF173567FD3C4FEE95B70B1A0D9B7503794ED81A7AA4861C5F1CEA72A26A2452544DA73AF7D57E095F3A54DFF82A5[0m

D (99387) HK_HomeKit: save_to_nvs > NVS SET STATUS: ESP_OK[0m

D (99388) HK_HomeKit: save_to_nvs > NVS COMMIT STATUS: ESP_OK[0m

D (99391) HK_HomeKit: processResult > SUB-TLV LENGTH: 13, DATA: 0208DF725617D9CB5A0D030100[0m

D (99398) HK_HomeKit: processResult > TLV LENGTH: 15, DATA: 050D0208DF725617D9CB5A0D030100[0m

[0;32mI (99407) HK_HomeKit: 05 0d 02 08 df 72 56 17 d9 cb 5a 0d 03 01 00 [0m
```

alpdmrel, 2024-08-17:
Tried updateing to iOS 18.1 beta 2 but same problem exists. Even removed and added again after the update. 

Also tried unlocking from the home app but it  switches to `Unlocking...` but stays there until cliked again and returns to `Locked`, same for the reverse stays at `Locking...`. The states `Unlocking` and `Locking...` is reflected to mqtt.

If `Unlocked` or `Locked` through home assistant then it works. 

These are the logs during Locking & Unlocking.

```
[0;32mI (5002424) LockMechanism: update > New LockState=1, Current LockState=0[0m

[0;32mI (5005300) LockMechanism: update > New LockState=0, Current LockState=0[0m

[0;32mI (5069593) LockMechanism: update > New LockState=1, Current LockState=0[0m

```

rednblkx, 2024-08-17:
Like i said, issue is that it's lacking the key that the iphone should be providing but looking at the logs, it is not, unfortunately i don't have any spare device to try on the Beta but i will be updating to iOS 18 once it's officially released to try and diagnose this.

The unlocking behaviour is intended as it's waiting for external feedback from you through MQTT, interacting with the lock from within HASS does exactly that.

alpdmrel, 2024-08-17:
Thank you anyway for your help. I will try to understand and try to fix it maybe if I can manage. :D

adukale, 2024-08-18:
FYI I have tried with iOS 18's new home architecture as well, the issue persists.

adukale, 2024-08-19:
> Tried updateing to iOS 18.1 beta 2 but same problem exists. Even removed and added again after the update.
>
> Also tried unlocking from the home app but it switches to `Unlocking...` but stays there [...]

Can you try setting Hardware Lock GPIO pin in Misc settings? I entered an available GPIO pin in this setting and after reboot the lock and unlock is working fine. Unlocking from home app updates the state as unclocked and then locking results in locking.
I also tested adding an LED on GPIO pin, the LED is updating as per the lock state.

rednblkx, 2024-08-21:
yes, that "solves" the lock state but the homekey functionality would still be lacking which makes the whole thing somewhat useless.

adukale, 2024-08-21:
For me the functionality is also working using iPhone as well as Apple Watch, as soon as the iPhone/watch comes in the range of NFC antena the Homekey gets validated and lock state gets updated in Home app.

alpdmrel, 2024-08-25:
@adukale which version of iOS 18 are you on and what is the model of your phone?

adukale, 2024-08-26:
I was on iOS 18.0 Developer beta 6 - iPhone 13 Pro Max when I added the comment. Today also I checked and It working for me, I am on iOS 18.0 (22A5346a) Developer beta 7 - iPhone 13 Pro Max.

alpdmrel, 2024-08-26:
Only difference seems to be the phone model. I have the 15 pro But I dont know that can affect anything?

rednblkx, 2024-08-26:
@adukale did you upgrade to iOS 18 with the device already paired or did you paired it on iOS 18?

adukale, 2024-08-27:
> Only difference seems to be the phone model. I have the 15 pro But I dont know that can affect anything? Have you upgraded to the new apple home version?

No, I am on old Apple Home version as I don't have homehub and new architecture requires homehub to add, share accessories.

adukale, 2024-08-27:
> @adukale did you upgrade to iOS 18 with the device already paired or did you paired it on iOS 18?

I paired the accessory on iOS 18.0 beta.

alpdmrel, 2024-08-27:
@adukale I don't think it requires homhub to add devices. But then probly it is a problem with new apple home architecture.

alpdmrel, 2024-08-27:
Okay, It finnaly works. I reverted back to the old homekit architecture with the help of this thread:
https://discussions.apple.com/thread/254934792?answerId=259247185022&sortBy=rank#259247185022
What I did was I installed the certificate from the post and removed my home then recreated a new apple home. 
And now it works. I get the state change through MQTT and also the animation on the phone goes through.

closed 2024-09-12 by rednblkx