rednblkx/HomeKey-ESP32 #115: Not sure if its working correctly...

rhamblen, 2024-08-23:
[image]

firstly I dont see a reader ID and reader GUD.  is that normal???

second, 
[image]
when I put a generic tag to the reader, I get an error and no tag ID.  To be honest that doesnt surprise me!

Ill do a separate message about the other issue I have that I cant see anywhere else.

rednblkx, 2024-08-23:
Lack of ID is not normal at least, those would be assigned during pairing but that would be due to issue #116 that you opened

The nfc error can be ignored, it's just because i'm sending a homekey command to it and the card responds with an error as indicated by the last message.
The tag id is only published to the mqtt topic "hk auth", it's only printed on console when it is set to debug level.

rhamblen, 2024-08-23:
[image]

that makes sense and fixing the pairing did then get it working correctly.

closed 2024-08-23 by rhamblen