rednblkx/HomeKey-ESP32 #116: Struggling to get a homekey on the iphone

rhamblen, 2024-08-23:
So Ive got the ESP scanner (I think) working.
I;ve got it recognised in the homekit device in home assistant.
[image]
when I lock and unlock the HK, I see the terminal doing stuff!!! ie requesting a change in state.

I've got homebridge on Home Assistant working and the lock is recognised on the iphone.

When I change the lock status on the iphone or on Home assistant, the changes flow through between devices and show on the terminal of the ESP HK.

BUT what I cant do is get the homekey to be issues on the iphone so I can check that NFC is working.  Nor can I work out how to set the rules for any authentication even if not a homekey.

Other than that, this looks like a solid option, and as you said elsewhere, you can use automation to unlock a door. (in my case a nuki)
Currently I use a tag and set a shortcut on the iphone to open the lock.  BUT its clumsy as I need to wait to connect to my local internet and I might as well go into the app.

rhamblen, 2024-08-23:
OK so I worked it out after a bacon sandwich!!!
Dont add via homebridge, add it directly to the phone!  
I was adding it the wrong way!!!!

rednblkx, 2024-08-23:
Hello! :wave: 

Homekey contains specific characteristics to be used by the Home App that Home Assistant doesn't know to forward to homebridge, it would just be a lock.

Also, if you wish to integrate it with HASS, the project already does that through MQTT Discovery, let me know if you need any help.

rhamblen, 2024-08-23:
got it all working... just needed that bacon butty and an few hours playing!!! 
cool piece of work... now need to 3D print an enclosure!

closed 2024-08-23 by rhamblen