rednblkx/HomeKey-ESP32 #120: Hardware question

darkside9009, 2024-08-28:
Hello together,

I did a nice small hardware, which include ESP32 and PN5321 chip. It's working well, but as we know the old PN5321 has a very short distance (there are some antenna tuning, but still not more then 4-5cm distance is possible)
I checked the PN5180, which is working with 10-15cm distance, the question is now, if there is any possibility to support also the PN5180 with your project? I could also create a new hardware (all in one)
[image]
[image]

mgoeller, 2024-08-28:
I was planning something very similar, could you please contact me via email at "mg (at) atios.ch". Thank you

rednblkx, 2024-08-28:
Hello!

It could be interesting to have that integrated and I do know that some people have been asking about longer ranger. I was already looking to potentially integrate a PN7160 to accommodate a battery but the PN5180 also looks like a good option.

Don’t like that I couldn’t really find a board with a small footprint like the ones from Elechouse as most people would be buying those since buying a custom board is more expensive.

A member of our discord server did an all in one board with the 532 but with an external antenna and is available on the PCBWay thingy but it’s expensive, pcb + assembly, goes to about 80$ and thats with a $20ish new user discount.

Anyway, like I said it would still be interesting to add support for that, I’ll see to maybe get a board from Ali and test it out, thanks for reaching out.

darkside9009, 2024-08-28:
Hi,
if there is a demand, we can offer this solution for everybody (the cost should be around 35 €, directly on Amazon) We have our own production here in Germany, we are also able to produce smaller quantities.
It would be great to integrate the PN7160. If you want, I can get a eval board for you and later redesign my board for PN7160.
(https://www.digikey.de/de/products/detail/nxp-usa-inc/568-OM27160B1-ND/15650131?utm_adgroup=&utm_source=google&utm_medium=cpc&utm_campaign=PMax%20Shopping_Product_Medium%20ROAS&utm_term=&productid=15650131&utm_content=&utm_id=go_cmp-20179585907_adg-_ad-__dev-c_ext-_prd-15650131_sig-CjwKCAjwlbu2BhA3EiwA3yXyu2hk_KftUXoFJF4hVByIjrdo3GdWexi-poyXtOSk3edFjyGJtjW3VxoCEuQQAvD_BwE&gad_source=1&gbraid=0AAAAADrbLlhSzkOlh3vWBqOpOMKj9m8AM&gclid=CjwKCAjwlbu2BhA3EiwA3yXyu2hk_KftUXoFJF4hVByIjrdo3GdWexi-poyXtOSk3edFjyGJtjW3VxoCEuQQAvD_BwE)

rednblkx, 2024-08-28:
> Hi, if there is a demand, we can offer this solution for everybody (the cost should be around 35 €, directly on Amazon) We have our own production here in Germany, we are also able to produce [...]

That's amazing, not sure if there would be a lot demand but we'll survey when we get there. At the moment i'm handling other things, not sure when i'll get to it, but it's something i want to add at some point. An eval board would be great to get started but i'm just too paranoid of people shipping me things.

I'll keep this issue open and it will be updated whenever there's progress on it.

darkside9009, 2024-08-28:
Hi,
I fully understand you. But you did a great project and I like to support people, they are loving what they are doing.
You don't need to be paranoid, just let me know. This is the minimum I can do to support you and your really great project.

jk2g, 2024-08-29:
eu compraria a placa pronta !!!!

rednblkx, 2024-08-29:
@darkside9009 we can talk more over email if you want

darkside9009, 2024-08-30:
> I was planning something very similar, could you please contact me via email at "mg (at) atios.ch". Thank you

did you got my email?

darkside9009, 2024-08-30:
> @darkside9009 we can talk more over email if you want

yes, no problem, I sent you already a email.

doerner, 2024-09-10:
I would suggest to add the UWB capability of the DWM3000 which might help for the wireless addition in the upcoming aliro standard.

doerner, 2024-09-10:
... and you really might consider to use the fastest chip in the ESP family to speed up the authorisation process.

darkside9009, 2024-10-11:
Hello all together,
if someone don't like or is not able to solder or connect all the wires to the boards, here is a link to buy a ready hardware which is directly working with this project here. Many thanks to the author :-)

https://casmo.info/product-details/?product=17

doerner, 2024-10-11:
Would it be possible to attach an external antenna for NFC or/ and for wifi?RegardsJan-Hendrik

https://casmo.info/product-details/?product=17

darkside9009, 2024-10-12:
Hi,
yes, this is possible, but its depending which antenna and the distance between antenna and the PN5321.

darkside9009, 2024-10-24:
Hi,

here we are. I have also included a 3d step file.
[CASmo-NFC_PCB.step.zip](https://github.com/user-attachments/files/17513360/CASmo-NFC_PCB.step.zip)
[CASmo-NFC.pdf](https://github.com/user-attachments/files/17513361/CASmo-NFC.pdf)

planecore, 2024-10-25:
> Hi,
> 
> here we are. I have also included a 3d step file. [CASmo-NFC_PCB.step.zip](https://github.com/user-attachments/files/17513360/CASmo-NFC_PCB.step.zip) [CASmo-NFC.pdf](https://github.com/user-attachments/files/17513361/CASmo-NFC.pdf)

Thanks! Sadly my order was cancelled. Oh well

darkside9009, 2024-10-25:
Hi,

sorry for that, we are checking with Amazon, why they canceled it. Please rebuy it, I will take care about your order.

dvd2000k, 2024-11-19:
Not available anymore on amazon?

darkside9009, 2024-11-19:
Should be available from tomorrow

schmidt-daniel, 2025-01-03:
Just wanted to order a board, but it's not available on Amazon currently. Will it come back eventually?

darkside9009, 2025-01-03:
We have closed during the Christmas, we will be back from 06.Jan.2025

electrontube, 2025-02-14:
This is a great board for the project, but unfortunately I don't have the space on the door module to install the board. Therefore, I have a suggestion for improvement: to provide the board with a predetermined breaking point between the ESP and RFID and to provide a JST connector on both sides of the separation point.
This means that anyone who has enough space can use the board as before, and if installation space is tight, it can be split and connected without soldering for the user.

darkside9009, 2025-02-15:
I can create a small hardware with Antenna and you can use it with our CASmo-S1 ( https://casmo.info/shop/casmo-s1-2#attr= ) , which is a standalone ESP32 with connectors and 2 x relays inside.

electrontube, 2025-02-16:
> I can create a small hardware with Antenna and you can use it with our CASmo-S1 ( https://casmo.info/shop/casmo-s1-2#attr= ) , which is a standalone ESP32 with connectors and 2 x relays inside.

Thanks for the quick response and your offer. 
CASmo-S1 would increase the footprint for this project beyond limits, since a tiny ESP32C3 Super Mini would do the job. 
What is currently missing, is a reliable PN532 module with a decent reading distance and  a small footprint. This in combination with some adapter PCBs for widely and cheaply available standard ESP Boards would certainly be a self selling product. 
Add some standard (i.e. JST) connectors with as small as possible spacing and we are all happy.

DuschdrBabbe, 2025-03-13:
@darkside9009 Thanks for your PCB! I have ordered one from Amazon! 
I am not good at 3D modeling. Can you also provide an STL file for a case? 
Or does anybody have an STL for a case for this cool little device? 

Thank you, Pascal.