rednblkx/HomeKey-ESP32 #133: Cannot access web UI after OTA update to 0.4

pnuding, 2024-10-08:
Installed release 0.4 via OTA on a device that previously had 0.3 on it
-espota reports success both for firmware and littlefs
-device is on wifi and responds to ping
-homekit sees it normally
-homekey authentications work and report on MQTT correctly
-Web UI is inaccessible from safari (cannot parse response) and firefox (error connecting) 
-tried power cycling, made no difference

pnuding, 2024-10-08:
tried on second device - same behaviour there

pnuding, 2024-10-08:
Tried erasing flash and flashing via USB. Shows same issue.

011V32, 2024-10-08:
Did u try to delete the cache and cookies?

rednblkx, 2024-10-08:
Hi!

Yeah, there is an issue with Safari on macOS Ventura and higher and iOS 18.

Other browsers should be working just fine.

Error connecting on Firefox would mean it’s not the right address though the fact that safari errors out with “cannot parse response” means it is the right ip address.

Perhaps try re-uploading the `littlefs.bin`, shouldn’t make a difference since you are updating but not the first time something crazy happens.

pnuding, 2024-10-08:
Cleared cache and it now works on Firefox. I also found an older Macbook running Monterey and there it works with Safari. 
Safari on Sequoia and iOS 18 still don't work even after clearing caches

rednblkx, 2024-10-08:
That’s great!

Ventura is oldest I tested as I’ve already had a VM and had some hunches that something got introduced in Ventura that followed up to Sequoia and your comment confirms indeed there is something between Monterey and Ventura, so thanks for that! Hopefully I’ll be able to figure what is it that Safari wants 😝

closed 2024-10-08 by rednblkx

sgruby, 2024-10-30:
I think the issue is that the HTML doesn't properly validate. I tried to monitor it using Uptime Kuma and it errors out indicating that it isn't a Safari issue.