rednblkx/HomeKey-ESP32 #138: Ip connection lost

ves1820, 2024-10-22:
my homekey (0.4.1) does lost the connection to WLAN-AP after some hours but show an ip in status on serial line.

> *** HomeSpan Status ***
> IP Address:        192.168.0.74

is it possible to reconnect on connection errors?

> E (75924662) esp-tls: [sock=50] select() timeout
> E (75924663) TRANSPORT_BASE: Failed to open a new connection: 32774
> E (75924663) MQTT_CLIENT: Error transport connect

rednblkx, 2024-10-31:
Are you able to ping the ip?

I believe it might be retaining the ip until next connection, haven’t looked at what HomeSpan or Arduino-ESP32 does on that front but the error you see from MQTT is the client trying to reconnect but there doesn’t seem to be anything to connect to.

ves1820, 2024-10-31:
The Broker is pingable the homespan not.
I had it running since late april this year until the end of september it was running without any problems. than i updated ...

rednblkx, 2024-11-03:
Well…I can assure you I haven’t made any changes that would affect the connectivity.

The issue lies on the reboot, sometimes the WiFi gets unstable after reboot. It usually helps to unplug it from power and plug it back in but I figure you already did that.

On that note, how are you powering the esp?

Also please take a look at the wifi rssi(on the console or the webui) once it connects, should be somewhere between -30 and -70 for a good signal

ves1820, 2024-11-03:
i had two of them working one right next to the AP
testet some power sources without success. 
then changed the esp and back again no luck.
It runs more stable since i changed to an older framework:
`framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#release/v2.x`
Maybe i have to go further back in time