rednblkx/HomeKey-ESP32 #140: HK Momentary State does not work

gimprota76, 2024-10-25:
HK Momentary State does not work, the padlock state does not return to lock after the set time
[image]

I have to open a door for now I'm using it without an RFID reader (coming soon) but from homekit when I press unlock after 5 seconds it doesn't return to the locked state furthermore the relay on gpio16 is always active
firmware version: v0.4.1
Thank you

rednblkx, 2024-10-31:
Hi!

The Momentary State is only used for HomeKey, I know it’s confusingly placed, didn’t figured out a better arrangement at the time.

It’s on my todo list to add a separate option for HomeKit.

gimprota76, 2024-10-31:
I wonder if you can implement the bulk return function "Momentary state" also from homekit app thanks!!!

rednblkx, 2024-11-20:
Fixed in [v0.5](https://github.com/rednblkx/HomeKey-ESP32/releases/tag/v0.5)

closed 2024-11-20 by rednblkx