rednblkx/HomeKey-ESP32 #141: wiring esp32s3

bosu1787, 2024-10-26:
hello i need wiring diagram to esp32s3. thx

bosu1787, 2024-10-26:
im getting allways :

E (5690) NFC_SETUP: Didn't find PN53x board

rednblkx, 2024-11-03:
Yeah, those should be the right pins.

The aforementioned error has been prevalent and usually occurs due to connectivity issues (bad wires, bad contacts, long wires, etc.) or even a bad nfc board or IC(possibly not genuine IC).

Could you post logs from the start of the boot? to see if the right pins are initialized.

bosu1787, 2024-11-03:
<spoiler>
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0xe68
load:0x403c9700,len:0xa14
load:0x403cc700,len:0x2d7c
entry 0x403c98d4
I (309) cpu_start: Multicore app
I (309) cpu_start: Pro cpu up.
I (309) cpu_start: Starting app cpu, entry point is 0x40376788
I (0) cpu_start: App cpu up.
I (327) cpu_start: Pro cpu start user code
I (327) cpu_start: cpu freq: 160000000
I (327) cpu_start: Application information:
I (329) cpu_start: Project name:     HomeKey-ESP32
I (335) cpu_start: App version:      v0.4.1
I (340) cpu_start: Compile time:     Oct 19 2024 17:04:41
I (346) cpu_start: ELF file SHA256:  4183b9e8041cc903...
I (352) cpu_start: ESP-IDF:          4.4.7
I (357) cpu_start: Min chip rev:     v0.0
I (361) cpu_start: Max chip rev:     v0.99
I (366) cpu_start: Chip rev:         v0.2
I (371) heap_init: Initializing. RAM available for dynamic allocation:
I (378) heap_init: At 3FCA28C8 len 00046E48 (283 KiB): D/IRAM
I (384) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (391) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (397) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (405) spi_flash: detected chip: generic
I (409) spi_flash: flash io: dio
W (412) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (450) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (451) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (458) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (468) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (478) system_api: Base MAC address is not set
I (482) system_api: read default base MAC address from EFUSE
I (489) sleep: Configure to isolate all GPIO pins in sleep state
I (495) sleep: Enable automatic switching of GPIO sleep configuration
I (502) cpu_start: Starting schedule2 oI (38) SETUP: setup > Reader Data loaded from NVS
I (52) SETUP: setup > MQTT Config loaded from NVS
I (63) SETUP: setup > Misc Config loaded from NVS
Listing directory: /
  FILE: actions.html    SIZE: 11534
  DIR : assets
  FILE: index.html      SIZE: 5790
  FILE: info.html       SIZE: 202
  FILE: misc.html       SIZE: 8623
  FILE: mqtt.html       SIZE: 10232
I (112) SETUP: setup > READER GROUP ID (8): 7642AAF012EE189E
I (112) SETUP: setup > READER UNIQUE ID (8): E31ACB8F1E775C31
I (116) SETUP: setup > HOMEKEY ISSUERS: 5
</spoiler>

bosu1787, 2024-11-06:
today i recived the esp32c3 and is working... so something is wrong with the flash for esp32s3 i suspect the gpio pins are wrong.