rednblkx/HomeKey-ESP32 #142: HTML in index.html doesn't validate

sgruby, 2024-10-30:
Using the validator at https://validator.w3.org/, index.html doesn't validate. This may be part of the issue why Safari can't load the main page.

rednblkx, 2024-10-31:
Hi!

yes, I know, the pages haven’t yet been made up to spec but that won’t prevent it from loading as it works fine when serving from a local http server.

this means there’s either something going on with ESPAsyncWebServer or i misconfigured it. Right now this is low priority since it only affects Safari. If you have any suggestions, let me know and also PRs are welcomed.

sgruby, 2024-10-31:
Thanks for the reply! I also noticed that using Uptime Kuma to monitor it, it gives me a "Parse Error: Invalid character in chunk size". Maybe that is related? I'll keep digging and see if I can pinpoint the loading failure for Safari and Uptime Kuma.

rednblkx, 2024-11-20:
Hi!

I've made some changes and Safari seems to be working now, please try the latest [CI build](https://github.com/rednblkx/HomeKey-ESP32/actions/runs/11896751048) and let me know, thanks.

sgruby, 2024-11-20:
Your release (0.5) appears to work, thanks!

closed 2024-11-20 by sgruby