rednblkx/HomeKey-ESP32 #143: A new tag with every scan

Astromechdroide, 2024-10-31:
Hello everyone, first of all, thank you very much for this great work. 
But I need your help. 
Every time I scan, the Apple ID and the device ID are correctly recognized, but an additional tag is also created in Home Assistant. Unfortunately, this happens every time. Do I need to configure anything else?

rednblkx, 2024-10-31:
Hi!

Just to be clear, are you saying there is a third tag appearing in HASS?

If that is the case then the NFC Antenna might not be in tune and is more prone to interferences. Unfortunately there’s nothing you can do without the expertise(which I also don’t have sadly). 

I don’t have yet a link to recommend for buy but you can have a look at the integrated boards designed by the community as outlined in the “[Prerequisites](https://github.com/rednblkx/HomeKey-ESP32/wiki/Prerequisites)” chapter on the wiki and their respective links to buy(I have no affiliation).

Astromechdroide, 2024-10-31:
Hi!

Thanks for the quick reply.

I checked all the cables and the dimmer switches again, everything is OK.

Then I tried to repeat the whole thing.
A new TAG is always created if the key card is not recognized correctly, i.e. there is a reading error.
When this error occurs, as in the picture.
[image]

Astromechdroide, 2024-11-02:
Can I disable the creation of tags on the ESP32?

rednblkx, 2024-11-03:
Unfortunately not, due to the way the integration with HASS works, anything that gets published to the configured MQTT topic is then added as tag or just triggers the tag if already present.

 I’ll be adding an option to ignore NFC tags.

Astromechdroide, 2024-11-04:
Danke für die Information. Dann freue ich mich auf ein Update. ;-)

rednblkx, 2024-11-20:
Added in [v0.5](https://github.com/rednblkx/HomeKey-ESP32/releases/tag/v0.5)

closed 2024-11-20 by rednblkx