rednblkx/HomeKey-ESP32 #148: One ESP with 2 PN532 NFC Modules with different action for each reader?

misc2000, 2024-11-09:
hello,
I use you project on 2 doors and it works perfect with the same homekey for all our apple devices for all family members !
But because I also have an alarm system with a KNX interface that is also integrated in Home Assistant I also want to use Homekey to Arm/disarm the alarm system to be able to retire my old system that arm/disarm this system.

Currently my alarm system uses a own token system.
And with that you unlock the door when you hold the token for a short time and arm the alarm system by holding the token for a long time to the reader.
If this is anyhow also possible with HomeKey it would be perfect.
But I thinks this 2 stats (short/long reading ) are not possible for Apple Homekey.

My idea now is to get with a workaround something similar.
I would connect 2 NFC Modules to the same ESP and use one to unlock the door and the second one to Arm/disarm the alarm system. With 2 ESP (one for each Reader its working but I have some space limitations and with only 1 ESP it would be much nicer)
And I think there should be some more users out there that also love to have 2 different actions at one entrance point depending to what reader you hold the iPhone/Watch.
Of corse I am also open to any other solution that helps me to manage my problem...
Thanks
Michael

rednblkx, 2024-11-20:
Hi!

I think this kind of interaction would be interesting and currently is not possible without using two separate ESP32 devices
 but tell you what, because i like this i'll keep it in mind as a potential future feature.
The workaround you mentioned would not be possible unfortunately as it cannot talk with two devices at the same time, the SPI protocol supports using the same bus with multiple devices but only one at a time and requires a software implementation for this which is not currently present.

misc2000, 2024-11-21:
Thanks for the update. If any help from my side is needed let me know...