rednblkx/HomeKey-ESP32 #151: After Reflashing the esp32 I can't add it to HomeKit Anymore

343max, 2024-11-10:
After erasing the flash and reflashing the esp32 I can't add the HomeKey anymore to the Home App.

After entering the setup code the process will just time out after a while. The error message reads "device can't be reached".

[image]

misc2000, 2024-11-10:
I have had a similar issue some days ago.
Did you already try to reset the HomeKit ID in the web config interface of the ESP?

rednblkx, 2024-11-20:
If you erased the flash make sure it's reconnected to wifi.
If you still have issues, hook it to a serial terminal, use the `L2` command, try to add it to Home and post the logs (make sure to truncate the data).

343max, 2024-11-21:
Sorry, forgot to close this. What seems to have fixed it was to rename the esp32 HomeKey so it was different from the name it had before.

closed 2024-11-21 by 343max