rednblkx/HomeKey-ESP32 #152: NFC module seems to be disconnected and does not work anymore

drstil, 2024-11-10:
Since I have done the update to version 0.4.1 (from version 0.3.1 or 0.3.11) I have issues with NFC module becomes not available anymore and only a restart (of HomeKey-ESP32) could make the scanning available again. This happens after a lot of hours or sometime after few days, but did not find any reproducible for this. I also checked the serial logs, but nothing really reproducible is seen. Once I had the following in the logs:

```
...
(12548969) MAIN: nfc thread entry > *** PASSIVE TARGET DETECTED ***
(12549167) HKAuthCtx: HKAuthenticationContext > Initialization Time: 171 ms
(12549355) HKAuthCtx: authenticate > Endpoint authenticated, transaction took 187 ms 
(12549367) MAIN: nfc thread entry › Total Time (detection→>auth=>gpio->mgtt): 396 ms
(12550145) MAIN: nfc thread entry > *** PASSIVE TARGET DETECTED ***
(12550340) HKAuthCtx: HKAuthenticationContext › Initialization Time: 169 ms
(12550553) PN532: Status code indicates an error
(12550554) HKAuthCtx: authenticate › Response not valid, something went wrong!
(12550556) MAIN: nfc_thread_entry > We got status FlowFailed, matt untouched!
(13594472) MAIN: nfc thread entry > *** PASSIVE TARGET DETECTED ***
(13594480) PN532 SPI: Received Error Frame - TFI: 7f CMD:81
(13594481) PN532: STATUS NFC: -5
(13594481) MAIN: nfc_thread_entry > Invalid Response, probably not Homekey, publishing target's UID
(13956020) PN532_SPI: PN532:: INVALID HEADER
```

I also tried with higher log levels for both HomeSpan and HomeKey, but could not get anything.

Not available means, that the HomeKey web interface is reachable, but when scanning a nfc tag, nothing is happening (in the serial logs). 
I tried different NFC modules (all same manufacturer (Elechouse)), but all lead to same issue. I think this is all information I could give. 

Any ideas, what could cause the issue?

daranto, 2025-01-17:
I think i have the same issue. Restart of the esp solves the issue.

drstil, 2025-01-20:
I also tried with v0.5 but same issue here. 

I could not easily use ESPHome version of this software, as there is a known issue with `esp-idf` versions prior to 5.3.2 when using ethernet connection: https://github.com/espressif/esp-idf/issues/14265. And I use WT32 ETH01. 

I did a lot of research and code adaptions. Added a periodic "ping" to PN532 module with `nfc->getFirmwareVersion()` did not really change anything. Also added a reinitialization of PN532 module, but as there is no real way to know if the PN532 module is unavailable (as it really seems to hang up), this also did not help. 
More debugging lead me to increasing the time interval in `nfc_thread_entry()` at the end of the `while (1) { ... }` from 50 to 200 ms (170 was definitely not enough), which probably helped. I will keep you updated.

segdy, 2025-09-13:
Did you guys ever figured out more?

I had a perfectly working system and was stupid enough to re-flash it. Now when I go back (exactly identical devices, hardware, power supplies, HomeKey-ESP32 version) I am having a [similar issue](https://github.com/rednblkx/HomeKey-ESP32/issues/202). However, for me, it's not just sometimes, it's always. It never scans but gives a similar log message. Restarting ESP, phone etc. does not help