rednblkx/HomeKey-ESP32 #155: [ERROR] Template variable warning: 'dict object' has no attribute 'uid' when rendering '{{ value_json.uid }}'

isibizi, 2024-11-18:
Hello, everytime when i open my door, i have theis error message in my log. Does anyone what know it is?

Logger: homeassistant.helpers.template
Quelle: helpers/template.py:2748
Erstmals aufgetreten: 09:39:35 (2 Vorkommnisse)
Zuletzt protokolliert: 10:49:52

Template variable warning: 'dict object' has no attribute 'uid' when rendering '{{ value_json.uid }}'

rednblkx, 2024-11-20:
Hello!

Yeah, that is due to the tag configuration for nfc tags. If an NFC Tag is detected, it will publish UID of that tag to the same MQTT Topic as for homekey just with a different data format hence the "error". I've already added a option in the upcoming version to ignore the nfc tags.

isibizi, 2024-11-20:
Thx for your reply and information.

rednblkx, 2024-11-20:
Added in [v0.5](https://github.com/rednblkx/HomeKey-ESP32/releases/tag/v0.5)

closed 2024-11-20 by rednblkx