rednblkx/HomeKey-ESP32 #171: Add Support For Regular NFC Tags

wqhqq1, 2025-01-26:
I have other members with android phone, so they can create nfc cards with a fixed tag ID, so I hope this project can allow me to add them to white list.

segdy, 2025-09-13:
Isn't this supported out of the box yet? It just creates a tracker for it...

But question @wqhqq1 , how did you manage to create an NFC card in Android with fixed ID? I was trying this myself recently but didn't find any app which would have allowed me that (only with rotating ones, which defeats the purpose)

segdy, 2025-09-14:
Ah got it. Because scanned tags show up as “trackers” and “last scanned” in home assistant. They could be easily used in an automation. That’s what I meant.