rednblkx/HomeKey-ESP32 #174: Wishlist

electrontube, 2025-02-17:
Thanks for this great project and all your effort!
To develop it even further I have a...

wishlist:

A - 1 or 2 GPIO inputs, that trigger publishing of a customizable MQTT topic 
B - 1 or 2 GPIO outputs than can be controlled by subscribing to an MQTT topic
C - MQTT with TLS
D - Provide AP to reconfigure WiFi if configured WiFi is unavailable
E - Option to configure WiFi hostname

rednblkx, 2025-02-18:
Hi there!

Thanks for the suggestions.

To answer briefly:

A & B - this has been requested a couple times and the thing is that i'm not aiming to make this into a standalone automation house but more of an endpoint that you can use in your automations, it's not particularly much to add but i'm trying not to give people the wrong that there's more to it than there is.
C - it's on plan to be added, probably on the next release
D - HomeSpan already has the option to attach a button(which can configured from the webui in the misc section) that can be used to turn on "Device Configuration" mode which among other things can turn on the AP, see [User Guide](https://github.com/HomeSpan/HomeSpan/blob/master/docs/UserGuide.md#device-configuration-mode)
E - Will be added, for now in the upcoming version i've standardized the naming scheme for the hostname, it will follow the format `HK-{serial_number}` and this will make it easier to access the device by just entering `HK-{serial_number}.local` since it already is using mDNS for HomeKit.

Cheers!

electrontube, 2025-02-19:
Thanks for your quick response!
A & B would help a lot, since I'm currently forced to run a second ESP just to get the state of a gate and to trigger a solenoid dependant on that state (via MQTT). That is really a waste of ressources and I would like to get rid of that second ESP, so it would be really appreciated if these Options find it's way into the firmware.
I'm not expecting any fancy logic here, just some GPIOs to be subscribed / published to via MQTT.

greetz!

electrontube, 2025-02-26:
wishlist:

_A - 1 or 2 GPIO inputs, that trigger publishing of a customizable MQTT topic
B - 1 or 2 GPIO outputs than can be controlled by subscribing to an MQTT topic
C - MQTT with TLS
D - Provide AP to reconfigure WiFi if configured WiFi is unavailable
E - Option to configure WiFi hostname_

F - OTA directly from WebUI