rednblkx/HomeKey-ESP32 #178: OTA - No response from device

electrontube, 2025-02-26:
Trying to update from 0.5 to 0.6 via OTA using espota.py I get the following:

Sending invitation to xxx.xxx.xxx.xxx 
Authenticating...OK
[ERROR]: No response from device

During this process the computer was on the same WiFi network and logged in to the WebUI of the device, so the device was definitely reachable.

rednblkx, 2025-02-27:
Is the esp32 on a different subnet? That's usually the issue, the way ArduinoOTA works requires the client to be accessible on the defined port(with espota.py), kinda annoying but i use it as it came with HomeSpan and spared me from doing the whole implementation but will be replaced with esp-idf's OTA process in the future.

electrontube, 2025-02-28:
The ESP is on the very same Network as the computer and there is no firewall or filtering in between

isibizi, 2025-05-20:
I Have the same issue

electrontube, 2025-06-10:
any update on tis issue?